e2002/yoradio #9: Distorted sound on I2S

dabu667, 2022-08-10:
First of all: very nice project!
I have a MAX98357a board connected but the sound is very loud and distorted - the sound behind the noise is barely noticeable. I tried the ESP32-audioI2S library example and the sound is OK using the same hardware setup so I suspect there is something wrong in your modified version of the library. Could you please update it to the latest version?

e2002, 2022-08-10:
Hey!
Could you send me a few things:
1. myoptions.h
2. Equalizer screenshot
3. Link to the station
4. If possible, record an example of distorted sound

PS: Thanks for the feedback.

dabu667, 2022-08-11:
Hi!

I put the files in Drive: [https://drive.google.com/drive/folders/1YqiXRTZCygMW0V_8pk8Wf3c3yd_dKTxQ?usp=sharing](url)
As you will see I added a SPI option for SH1106 but you can ignore that. I even compiled your code with the dummy display so I can exclude the possibility that the problem comes from the SPI communication.

I set the gain to minimum 3dB on MAX97357a but it does not have any effect.
I tried different speakers, but for testing I use this small crappy one because the coil gets hot if left connected for more than a few seconds.
I use many streams MP3, AAC, different bitrates but get the same result.

My guess (but I'm a noob) is that the wrong data is sent to the I2S interface. Could this be it: [https://forum.arduino.cc/t/esp32-i2s-bug-lsb-should-be-on-the-next-bclk/893936](url) ?

Anyway, as you can see the newest Audio.h library version works, so maybe you can update your code?

The stream I used for this test is 128kb mp3: [http://mp3.ffh.de/radioffh/hqlivestream.mp3](url) 

Thanks for replying, great project, nice responsive web interface.
Dan

e2002, 2022-08-11:
OK, tell me what version of Arduino and ESP32 board/platform are you using?
[image]

It would also be nice to see this screenshot:

[image]

dabu667, 2022-08-11:
Hi,
Board version:
[image]
Board config:
[image]
As you can see I have Arduino 1.8.12. I will ipdate to the latest.

e2002, 2022-08-11:
Please change the board model in sketch upload settings to "ESP32 Dev Module"
Partition Scheme "Minimal SPIFFS ...etc"

dabu667, 2022-08-11:
That worked!
[image]
I didn't think it would, but it did. Thank you!. Would you care to explain why it worked? Is it related to PSRAM?

e2002, 2022-08-11:
Not sure, maybe a bug in the ESP core for the WROOM-DA board

closed 2022-08-12 by dabu667