e2002/yoradio #10: OGG Vorbis playback

dabu667, 2022-08-12:
Is this project meant to play also OGG streams? It shows the title / artist info but there is no sound.
MP3 and AAC works.
I understand FLAC is not possible without PSRAM.

e2002, 2022-08-12:
Hello!
This project must support the same audio formats as ESP32-audioI2S.
Please send me links to streams.

dabu667, 2022-08-13:
Hi!
Below are the stream URLs and the serial output for each one (there is no sound):

[http://stream.radioreklama.bg/energy-90s.ogg](url)
_##AUDIO.INFO#: buffers freed, free Heap: 143052 bytes
##CLI.META#: [connecting]
> ##CLI.NAMESET#: 3 NRJ Bulgaria
##AUDIO.INFO#: buffers freed, free Heap: 146996 bytes
##AUDIO.INFO#: Connect to new host: "http://stream.radioreklama.bg/energy-90s.ogg"
##AUDIO.INFO#: Connect to "stream.radioreklama.bg" on port 80, extension "/energy-90s.ogg"
##AUDIO.INFO#: Connected to server in 262 ms
##AUDIO.INFO#: HTTP/1.0 200 OK
##AUDIO.INFO#: ContentType application/ogg found
##AUDIO.INFO#: date: Sat, 13 Aug 2022 05:04:26 GMT
##AUDIO.INFO#: ice-audio-info: channels=2;quality=0.2;samplerate=44100
##AUDIO.INFO#: icy-name: Energy 90s
##CLI.INFO#
##SYS.DATE#: 2022-08-13T08:04:26+03:00
##CLI.NAMESET#: 3 NRJ Bulgaria
##CLI.META#: Energy 90s
##CLI.VOL#: 63
##CLI.PLAYING#
> ##AUDIO.INFO#: server: Icecast 2.4.0-kh13
##AUDIO.INFO#: access-control-allow-origin: *
##AUDIO.INFO#: access-control-allow-headers: Origin, Accept, X-Requested-With, Content-Type, Icy-MetaData
##AUDIO.INFO#: access-control-allow-methods: GET, OPTIONS, HEAD
##AUDIO.INFO#: Switch to DATA, metaint is 0
##AUDIO.INFO#: stream ready
##AUDIO.INFO#: buffer filled in 2 ms
WebSocket client #1 connected from 192.168.1.171
##CLI.META#: Energy 90s
> _

[https://st01.sslstream.dlf.de/dlf/01/high/opus/stream.opus](url)
_##AUDIO.INFO#: buffers freed, free Heap: 147364 bytes
##CLI.META#: [connecting]
> ##CLI.NAMESET#: 7 Deutschlandfunk
##AUDIO.INFO#: buffers freed, free Heap: 145496 bytes
##AUDIO.INFO#: Connect to new host: "https://st01.sslstream.dlf.de/dlf/01/high/opus/stream.opus"
##AUDIO.INFO#: Connect to "st01.sslstream.dlf.de" on port 443, extension "/dlf/01/high/opus/stream.opus"
##AUDIO.INFO#: SSL has been established in 1156 ms, free Heap: 98500 bytes
##AUDIO.INFO#: HTTP/1.1 302 Found
##AUDIO.INFO#: date: Sat, 13 Aug 2022 05:08:15 GMT
##AUDIO.INFO#: content-length: 0
##AUDIO.INFO#: access-control-allow-methods: GET, OPTIONS, HEAD
##AUDIO.INFO#: access-control-allow-origin: *
##AUDIO.INFO#: redirect to new host "https://d141.rndfnk.com/ard/dlf/01/opus/64/stream.opus?cid=01FBPWZ12X2XN8SDSMBZ7X0ZTT&sid=2DHsMTUQg7Ksr55XIuWr7eLtRpa&token=3UlG96m2ia0NS_s0UmBjgiJXOvEW##AUDIO.INFO#: buffers freed, free Heap: 146912 bytes
##AUDIO.INFO#: Connect to new host: "https://d141.rndfnk.com/ard/dlf/01/opus/64/stream.opus?cid=01FBPWZ12X2XN8SDSMBZ7X0ZTT&sid=2DHsMTUQg7Ksr55XIuWr7eLtRpa&token=3UlG96m2ia0NS_s0UmBjgiJXOvEW##AUDIO.INFO#: Connect to "d141.rndfnk.com" on port 443, extension "/ard/dlf/01/opus/64/stream.opus?cid=01FBPWZ12X2XN8SDSMBZ7X0ZTT&sid=2DHsMTUQg7Ksr55XIuWr7eLtRpa&token=3UlG96m2ia0NS_s0UmBj##AUDIO.INFO#: SSL has been established in 1494 ms, free Heap: 101160 bytes
##AUDIO.INFO#: HTTP/1.1 200 OK
##AUDIO.INFO#: ContentType application/ogg found
##AUDIO.BITRATE#: 128000
##AUDIO.INFO#: ice-audio-info: ice-samplerate=44100;ice-bitrate=128;ice-channels=2
##AUDIO.INFO#: icy-name: Deutschlandfunk
##CLI.INFO#
##SYS.DATE#: 2022-08-13T08:08:17+03:00
##CLI.NAMESET#: 7 Deutschlandfunk
##CLI.META#: Deutschlandfunk
##CLI.VOL#: 63
##CLI.PLAYING#
> ##AUDIO.INFO#: icy-samplerate: 44100
##AUDIO.INFO#: server: rocketstreamingserver
##AUDIO.INFO#: date: Sat, 13 Aug 2022 05:08:17 GMT
##AUDIO.INFO#: access-control-allow-origin: *
##AUDIO.INFO#: access-control-expose-headers: Content-Type, icy-metaint, icy-br, ice-audio-info, icy-description, icy-genre, icy-name, icy-samplerate, icy-url
##AUDIO.INFO#: Switch to DATA, metaint is 16384
##AUDIO.INFO#: stream ready
##AUDIO.INFO#: buffer filled in 429 ms
_

[http://dancewave.online/dance.ogg](url)
_##AUDIO.INFO#: buffers freed, free Heap: 146324 bytes
##CLI.META#: [connecting]
> ##CLI.NAMESET#: 8 dance ogg 160
##AUDIO.INFO#: buffers freed, free Heap: 144304 bytes
##AUDIO.INFO#: Connect to new host: "http://dancewave.online/dance.ogg"
##AUDIO.INFO#: Connect to "dancewave.online" on port 80, extension "/dance.ogg"
##AUDIO.INFO#: Connected to server in 60 ms
##AUDIO.INFO#: HTTP/1.1 302 Moved Temporarily
##AUDIO.INFO#: server: nginx
##AUDIO.INFO#: date: Sat, 13 Aug 2022 05:07:11 GMT
##AUDIO.INFO#: content-length: 0
##AUDIO.INFO#: redirect to new host "http://stream.dancewave.online:8080/dance.ogg"
##AUDIO.INFO#: buffers freed, free Heap: 145816 bytes
##AUDIO.INFO#: Connect to new host: "http://stream.dancewave.online:8080/dance.ogg"
##AUDIO.INFO#: Connect to "stream.dancewave.online" on port 8080, extension "/dance.ogg"
##AUDIO.INFO#: Connected to server in 42 ms
##AUDIO.INFO#: HTTP/1.0 200 OK
##AUDIO.INFO#: ContentType application/ogg found
##AUDIO.INFO#: date: Sat, 13 Aug 2022 05:07:11 GMT
##AUDIO.BITRATE#: 160000
##AUDIO.INFO#: ice-audio-info: channels=2;bitrate=160;quality=5;samplerate=44100
##AUDIO.BITRATE#: 160000
##AUDIO.INFO#: icy-name: Dance Wave!
##CLI.INFO#
##SYS.DATE#: 2022-08-13T08:09:24+03:00
##CLI.NAMESET#: 8 dance ogg 160
##CLI.META#: Dance Wave!
##CLI.VOL#: 63
##CLI.PLAYING#
> ##AUDIO.INFO#: icy-sr:44100
##AUDIO.INFO#: icy-vbr:1
##AUDIO.INFO#: server: Icecast 2.4.0-kh15 (MSCP)
##AUDIO.INFO#: access-control-allow-origin: *
##AUDIO.INFO#: access-control-allow-headers: Origin, Accept, X-Requested-With, Content-Type, Icy-MetaData
##AUDIO.INFO#: access-control-allow-methods: GET, OPTIONS, SOURCE, PUT, HEAD, STATS
##AUDIO.INFO#: Switch to DATA, metaint is 0
##AUDIO.INFO#: stream ready
##AUDIO.INFO#: buffer filled in 3 ms
_

Thanks for looking into it

e2002, 2022-08-13:
Judging by the ESP32-audioI2S [source code](https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/src/Audio.cpp#L3736), the OGG Vorbis/Opus codec is currently not supported.

[image]

I'm using an older version of the library, but I don't think it has this support.

dabu667, 2022-08-16:
You are right - OGG is not supported even in the latest version of the ESP32-audioI2S lib. I should have checked that before posting here.

closed 2022-08-16 by dabu667