e2002/yoradio #11: This is a request, not an issue: auto-shutdown / sleep

dabu667, 2022-08-16:
Hi again,
Could you please implement a power save function: if on pause, after 10 minutes or so, disconnect from the wifi, put ESP to sleep, turn off LCD.
Wake up on any button. Is it possible? Thank you.

e2002, 2022-08-17:
Hello!
This feature can be implemented right now using the plugin.
Place the deepsleep.ino file in the root of the project and upload the sketch.

---
**deepsleep.ino**
```
#define SLEEP_DELAY     60  // 1 min
#define WAKEUP_PIN_1    GPIO_NUM_12
#define WAKEUP_LEVEL    LOW

Ticker deepSleepTicker;

void goToSleep(){
  if(BRIGHTNESS_PIN!=255) analogWrite(BRIGHTNESS_PIN, 0); /*  BRIGHTNESS_PIN added in v0.7.330  */
  esp_deep_sleep_start();
}

void yoradio_on_setup(){
  esp_sleep_enable_ext0_wakeup(WAKEUP_PIN_1, WAKEUP_LEVEL);
  deepSleepTicker.attach(SLEEP_DELAY, goToSleep);
}

void player_on_start_play(){
  deepSleepTicker.detach();
}

void player_on_stop_play(){
  deepSleepTicker.attach(SLEEP_DELAY, goToSleep);
}
```

dabu667, 2022-08-17:
Cool
Is GPIO12 the only pin that has external interrupts?
Is there a function being called at wake up so I can restore the brightness value?

e2002, 2022-08-17:
No, any RTC GPIO

[image]

https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

https://randomnerdtutorials.com/esp32-external-wake-up-deep-sleep/

e2002, 2022-08-17:
_> Is there a function being called at wake up so I can restore the brightness value?_
When waking up, the last saved brightness value is restored

dabu667, 2022-08-18:
ESP goes to sleep and wakes up as supposed, but I can't turn the display off. I have an SH1106 OLED and the brightness pin doesn't help me. I tried to clear it (being OLED it's almost like being turned off) like this:
`void goToSleep(){
  dsp.clearDsp();
  esp_deep_sleep_start();
}`
but the display only freezes before esp sleeps. Does it need a delay between the two commands?

dabu667, 2022-08-18:
It seemd that `dsp.clearDsp();` worked but then, just before sleep the display was updated in another task.
I figured out how to turn it off:
```
  dsp.oled_command(0xAE);        //display OFF
  uint8_t cmd[] = {0x8D, 0x10}; 
  dsp.oled_commandList(cmd, 2);  //disable charge pump
```

dabu667, 2022-08-19:
Thanks for helping me buid this:
[image]

closed 2022-08-19 by dabu667