e2002/yoradio #12: Error compiling version v0.7.330 and v0.7.355

dorinik, 2022-08-22:
Hello!
First of all, thanks for your work !
Unfortunately, starting with version v0.7.330 as well as in v0.7.355, the compilation no longer completes successfully with the error:
D:\yoradio-0.7.355\yoRadio/displayhandlers.ino:33: multiple definition of `getForecast()'; sketch\display.cpp.o:D:\yoradio-0.7.355\yoRadio/display.cpp:707: first defined here
collect2.exe: error: ld returned 1 exit status
exit status 1
I mention that version 0.7.017 compiles successfully.
I use the ILI9341 display.

vledenev, 2022-08-22:
I also use ILI9341 display and version 0.7.355. There are no installation problems. Check your environment and settings in myoptions.h file

dorinik, 2022-08-22:
I use the same settings in myoptions.h, both in v0.7.017 and v0.7.355 (v0.7.330).
I noticed that starting with v0.7.330, getForecast() is defined both in display.cpp and in displayhandlers.ino. (as the Arduino IDE also mentions in the error message)
In v0.7.017 it is only defined in displayhandlers.ino
[image]
[image]
[myoptions.zip](https://github.com/e2002/yoradio/files/9392607/myoptions.zip)
.

e2002, 2022-08-22:
@dorinik Since [v0.7.330](https://github.com/e2002/yoradio#v07330) the weather plugin has been integrated into the main code. 
displayhandlers.ino is no longer needed, please remove it from the project root directory.

PS: The examples/plugins directory contains only examples of how the user can interact with the main program code, for such an example, the displayhandlers.ino file is left there.

dorinik, 2022-08-23:
After removing displayhandlers.ino, compilation successful!
Thanks for the quick reply!

closed 2022-08-23 by dorinik