e2002/yoradio #13: ST7735 turns black immediately after boot

dabu667, 2022-09-07:
When the board starts the screen goes white for a second, then it briefly displays the eRadio logo and then it turns black.
I see your comment from myoptions.h mentioning that for color displays is needed more memory. I couldn't find
`xTaskCreateUniversal(_async_service_task, "async_tcp", 8192 * 2, NULL, 3, &_async_service_task_handle, CONFIG_ASYNC_TCP_RUNNING_CORE);`
instead I found in line 184:
`xTaskCreatePinnedToCore(_async_service_task, "async_tcp", 8192, NULL, 3, &_async_service_task_handle, ASYNCTCP_RUNNING_CORE);`
and i tried halving that but the screen still turns black after boot.
I should mention that the example from Adafruit ST7735 lib works ok.  In the documentation is mentioned that a very fast SPI speed could manifest like this. Are you using a different SPI freq then the default one?

dabu667, 2022-09-07:
I recompiled everything and now it works. Strange.
Now I have another problem: I have strange characters on the weather line. Could it be that the API key is not activated yet? I will update if anything changes.
[image]

vledenev, 2022-09-07:
Replace the file Arduino/libraries/Adafruit_GFX_Library/glcdfont.c with the file yoRadio/fonts/glcdfont.c for localization Adafruit_GFX

dabu667, 2022-09-07:
Thanks Vlad, I tried that but then it showed Cyrillic characters and then I understood it was about the fact that the text is in russian. I figured out that the weather API was returning results in russian language and also some predefined strings like "pressure" and "humidity" were in russian. in display.cpp.

closed 2022-09-07 by dabu667