e2002/yoradio #15: Request: bluetooth audio

dabu667, 2022-10-03:
Hi,

I came across this by accident [https://github.com/alexus2033/ESP32-A2DP](url)
Do you think it is possible to implement bluetooth audio? That is, to be able to switch from the web interface to a mode in which it works like a bluetooth speaker?
I know this is a lot to ask and I will close the issue regardless.

ESPNoodle, 2023-07-27:
I dont think the radio can work simultaneously, so if Wi-Fi on BT will not operate on esp32. (Shame!) ;)

stjames12, 2023-11-17:
Hello Guys, am I thinking about the bluetooth audio, as I read on the wiki it is possible to use this webraido as simple music player, play music from SD card without wifi. "When booting in AP mode, the PLAY button and the MODE button restart the ESP in SDCARD mode.", Maybe it is possible to restart the ESP in bluetooth mode as well?

e2002, 2025-09-16:
Hi, adding bluetooth support is not planned in the foreseeable future. I'm sorry.