e2002/yoradio #20: Library Adafruit_ST7735

motardu, 2022-12-10:
Hello!
First of all, thanks for your work !  I use Arduino IDE v.2.0.3 and Arduino core for the ESP32 v2.0.5   and yoradio-0.8.112 the 
 compilation is unsuccessfully with the error:.....\Adafruit_ST7735-master/Adafruit_ST7735.h:113:3: note: candidate:'Adafruit_ST7735::Adafruit_ST7735(uint8_t, uint8_t, uint8_t)'Adafruit_ST7735(uint8_t CS, uint8_t RS, uint8_t RST); ......
I changed the library with Adafruit_ST7735_and_ST7789_Library and  everything went well.
I post this  because it can help someone.

e2002, 2022-12-10:
@motardu Hey!
_> Adafruit_ST7735-master_
Libraries for displays should be installed through the library manager, as you did.
Thank You!
https://github.com/e2002/yoradio/wiki/How-to-flash Preparing#4

e2002, 2022-12-10:
PS: Arduino IDE versions higher than 1.8.19 are not currently supported (or there are restrictions)

closed 2022-12-10 by e2002