e2002/yoradio #23: Touchscreen issue

dabu667, 2023-02-10:
Hi,

I have this board: 2.8inch_ESP32-2432S028R
[image]
It uses XPT2046 as the touchscreen driver with the following connections too the ESP32:

```
#define XPT2046_IRQ		36
#define XPT2046_MOSI		32
#define XPT2046_MISO		39
#define XPT2046_CLK		25
#define XPT2046_CS		33
```
[image]

Any chance to make the touchscreen work?

e2002, 2023-02-10:
Hi
Here is the config of this touchscreen
```
#define TS_MODEL            TS_MODEL_XPT2046
#define TS_SPIPINS          25, 39, 32    /* SCK, MISO, MOSI */
#define TS_CS               33
```
https://github.com/e2002/yoradio/wiki/ESP32-TFT-Boards-configurations#esp32-2432s028r

dabu667, 2023-02-11:
So it was already documented. Thank you.
How would I display the stream type (MP3/AAC) above bitrate on screen?

e2002, 2023-02-12:
> So it was already documented. Thank you. How would I display the stream type (MP3/AAC) above bitrate on screen?

This feature is not currently supported.

closed 2023-02-13 by dabu667