e2002/yoradio #28: Next, Prev buttons

dabu667, 2023-03-20:
I wanted to have PREV, NEXT buttons that worked on single click.
For that I defined the buttons in **_myoprions.h_**:
`#define BTN_UP        36           /*  Prev, Move Up */
#define BTN_DOWN  39           /*  Next, Move Down */
`
but pressing them shows the stations list first. I just wanted to skip to the next previous station without showing the list like a normal radio does (i. e. the on in your car).

I modified the _**onBtnClick**_ function in _**controls.cpp**_ as below: 
`   case EVT_BTNUP:
    case EVT_BTNDOWN: {
        //if (DSP_MODEL == DSP_DUMMY) { 
          if (id == EVT_BTNUP) {
            player.next();
          } else {
            player.prev();
          }
        //} else {
        //  if (display.mode() == PLAYER) {
        //    display.putRequest(NEWMODE, STATIONS);
        //  }
        //  if (display.mode() == STATIONS) {
        //    controlsEvent(id == EVT_BTNDOWN);
        //  }
        //}
        break;
      }
`
 .. but I was wondering if there's a simpler way to do it.

e2002, 2023-03-20:
@dabu667 
_>> but I was wondering if there's a simpler way to do it._
No. It's just that I don't like navigating through the stations, getting into the delay caused by the connection time. If there is a display, moving through the list is much faster :)

dabu667, 2023-03-21:
You're probably right.

closed 2023-03-21 by dabu667