e2002/yoradio #29: web authentication

dabu667, 2023-03-20:
Is there a way to implement a simple password for accessing the web interface?

e2002, 2023-03-20:
No, this feature is currently not implemented.

dabu667, 2023-03-21:
Maybe a way to change the URL for the main page? For example <IP>/main.html?

memisibram, 2023-12-25:
Yes, this is possible in a simple way, it can only be applied to web interface files.
You will need to extract and re-archive the script.js.gz file.
Edit the script.js file.
find function onLoad(event)
and replace it with this
`function onLoad(event) {
	if(window.sessionStorage.getItem("success") == "yes") {
		initCheckboxes();
		initSliders();
		initButtons();
		initWebSocket();
	} else {
		if(prompt("Please enter your password","") == "pass12345") {
			window.sessionStorage.setItem("success", "yes");
			onLoad();
		} else {
			window.sessionStorage.clear();
			onLoad();
		}	
	}
}`
archive again and update files from web interface.

ciprisat, 2024-12-29:
is it possible to choose port 5000 not 80? the necessary settings can be made

e2002, 2025-09-15:
Basic HTTP authentication is supported since v0.9.530 https://github.com/e2002/yoradio#v09530