e2002/yoradio #32: Request: Fixed IP option for SSID

ESPNoodle, 2023-04-20:
Hi, Absolutely loving yoradio!! 

I have one wifi network which is has many set IP's, if I try to get the yoradio to connect to it, it will not as it is being assigned a DHCP address which clashes with a fixed IP. on that network. if it does connect it sits there just frozen.

Could there be away to add a fixed or DHCP button?

Big  thanks for all this amazing work. ;)

acoveca, 2023-09-18:
i used following Arduino Sketch to identify my ESP32s MAC Address and with this information i assigned a fixed IP-Address to
my yoRadio in my router.

/ Complete Instructions to Get and Change ESP MAC Address: https://RandomNerdTutorials.com/get-change-esp32-esp8266-mac-address-arduino/

#ifdef ESP32
  #include <WiFi.h>
#else
  #include <ESP8266WiFi.h>
#endif

void setup(){
  Serial.begin(115200);
  Serial.println();
  Serial.print("ESP Board MAC Address:  ");
  Serial.println(WiFi.macAddress());
}
 
void loop(){
}

ESPNoodle, 2023-09-20:
> i used following Arduino Sketch to identify my ESP32s MAC Address and with this information i assigned a fixed IP-Address to my yoRadio in my router.
>
> / Complete Instructions to Get and Change ESP [...]
/

acoveca, 2023-09-20:
i don't need to add anything to the Yoradio code, because i get always the same fixed IP address i have programmed on my router. In my case 192.168.1.120. Using those address on any browser, i get access to the web page of Yoradio, from my laptop, cellular, tablett, etc.