e2002/yoradio #33: Title and artist are not displayed.

hevet, 2023-04-24:
When there are Polish characters in the name of the artist or title, the artist and title are not displayed, SH1106 display.
Example station
http://stream.rcs.revma.com:80/cvswvmyewzzuv

hevet, 2023-04-28:
Dzięki za wyjaśnienie, spróbuję może coś mi się uda zrobić.

hevet, 2023-05-07:
@e2002 co by trzeba było zmienić, by wyświetlały się małe i wielkie litery, a nie tylko wielkie?

e2002, 2023-05-07:
@hevet 
W pliku yoRadio/src/displays/conf/display<TWÓJWYŚWIETLACZ>conf.h zmień parametr uppercase=false dla żądanego widżetu.

[image]

hevet, 2023-05-07:
e2002 dzięki :).
Swoją drogą mam już polskie znaki w tytułach.

hevet, 2023-05-08:
@e2002 Mam kolejne pytanie, jaki parametr odpowiada za szybkość przewijania tytułu czy wykonawcy oraz pogody i w którym pliku jest to zapisane. Dzięki