e2002/yoradio #34: Compatibility with esp32-2432s028 ?

echosmart792, 2023-04-25:
Hi. I managed to install yoRadio on the esp32-2432s028 board, on which I have an image on the display and sound on the integrated audio output with the help of the internal DAC. But I have a small problem. I can't manage to configure the touchscreen for yoRadio, even though I tried countless configurations with the help of ёRadio - simple myoptions.h generator. Can you make yoRadio fully functional on this board? I bought the board here: https://www.aliexpress.com/item/1005004502250619.html and I found documentation about the board here: https://macsbug.wordpress.com/2022/08/17/esp32-2432s028 /
I mention that this webradio: https://macsbug.wordpress.com/2022/08/20/web-radio-esp32-2432s028-i2s/ has a functional touchscreen on this board. Can yoRadio be adapted to work correctly with the screen and touchscreen and with audio on the internal DAC?
At the moment I have sound and image on the screen, but the image only shows the volume level and the IP to which I need to connect for the web interface. I will attach the link with the file with image and sound but without touchscreen: https://e2002.github.io/docs/myoptions-generator.html?r=71,1,2,5,8,19,42, 45,46,60,63&i=1,2,3,4,34&v=2,15,-1,255,21 I also attach an image of what appears on my screen: https://i.imgur.com/GWV11oG. jpeg

dabu667, 2023-04-25:
See this answer here: [](https://github.com/e2002/yoradio/issues/23)

echosmart792, 2023-04-25:
Thanks! Works great! 
My configuration with direct sound from the audio output is the following:

#ifndef myoptions_h
#define myoptions_h

#define DSP_MODEL           DSP_ILI9341
#define DSP_HSPI            true
#define LED_INVERT          true
#define TFT_CS              15
#define TFT_DC              2
#define TFT_RST             -1
#define LED_BUILTIN         17
#define BRIGHTNESS_PIN      21
#define I2S_INTERNAL			true
#define PLAYER_FORCE_MONO			true
//#define I2S_DOUT            22
//#define I2S_BCLK            16
//#define I2S_LRC             4
#define TS_MODEL            TS_MODEL_XPT2046
#define TS_SPIPINS          25, 39, 32    /* SCK, MISO, MOSI */
#define TS_CS               33
//#define IR_PIN              27
#define LIGHT_SENSOR        34
#define AUTOBACKLIGHT_MAX   1024
#endif

The tochscreen also works excellently!

closed 2023-04-25 by echosmart792