e2002/yoradio #38: Nie odtwarza żadnej stacji radiowej.

hevet, 2023-05-19:
@e2002  Po aktualizacji do najnowszej wersji 0.9.200, nie działa odtwarzanie radia, muzyka z SD działa prawidłowo. Usuwałem cała listę wgrywałem na nowo, czyściłem przeglądarkę, cały czas pokazuje komunikat "Error connecting to". Wszystko wgrane na czysto.

sp3rat, 2023-05-19:
Może ustawienia sieci się pozmieniały?

hevet, 2023-05-19:
Nie, wszystko wpisywałem na pewno dobrze, czas się wczytał, pogoda też działa.

e2002, 2023-05-19:
@hevet  
1. Wyślij mi log z Monitora szeregowego i plik playlist.csv.
Log należy skopiować po otwarciu Monitora szeregowego, naciśnięciu przycisku Wyczyść w oknie, które zostanie otwarte, naciśnięciu przycisku Reset na ESP, rozpoczęciu odtwarzania kilku stacji.
W tym oknie działają klawisze Ctrl-A, Ctrl-C
Szybkość portu musi być ustawiona na 115200

2. Jeśli otworzysz listę odtwarzania do edycji, co się tam znajdzie?

hevet, 2023-05-19:
lista stacji 
[playlist(7).csv](https://github.com/e2002/yoradio/files/11519907/playlist.7.csv)

hevet, 2023-05-19:
##CLI.VOL#: 47
##CLI.META#: [łączenie]
##ERROR#:	Request http:// failed!
##ERROR#:	Request http:// failed!
##CLI.META#: ёRadio
##ERROR#:	Error connecting to 
##CLI.META#: Error connecting to 
##ERROR#:	Error connecting to 
##CLI.META#: Error connecting to 
##CLI.INFO#
##SYS.DATE#: 2023-05-19T21:32:37+03:00
##CLI.NAMESET#: 0 ёRadio
##CLI.VOL#: 47
##CLI.STOPPED#
##CLI.META#: [łączenie]
##ERROR#:	Request http:// failed!
##ERROR#:	Request http:// failed!
##CLI.META#: ёRadio
##ERROR#:	Error connecting to 
##CLI.META#: Error connecting to 
##ERROR#:	Error connecting to 
##CLI.META#: Error connecting to 
##CLI.INFO#
##SYS.DATE#: 2023-05-19T21:32:38+03:00
##CLI.NAMESET#: 0 ёRadio
##CLI.VOL#: 47
##CLI.STOPPED#
##CLI.META#: [łączenie]
##ERROR#:	Request http:// failed!
##ERROR#:	Request http:// failed!
##CLI.META#: ёRadio
##ERROR#:	Error connecting to 
##CLI.META#: Error connecting to 
##ERROR#:	Error connecting to 
##CLI.META#: Error connecting to 
##CLI.INFO#
##SYS.DATE#: 2023-05-19T21:32:39+03:00
##CLI.NAMESET#: 0 ёRadio
##CLI.VOL#: 47
##CLI.STOPPED#

hevet, 2023-05-19:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
##[BOOT]#	SPIFFS mounted
##[BOOT]#	************************************************
##[BOOT]#	*               ёPadio v0.9.200                *
##[BOOT]#	************************************************
##[BOOT]#	------------------------------------------------
##[BOOT]#	arduino:	10819
##[BOOT]#	compiler:	8.4.0
##[BOOT]#	esp32core:	2.0.8
##[BOOT]#	chip:		model: ESP32-D0WDQ6 | rev: 1 | id: 16646532 | cores: 2 | psram: 0
##[BOOT]#	display:	1
##[BOOT]#	audio:		VS1053 (27, 14, 26, 12, false)
##[BOOT]#	audioinfo:	false
##[BOOT]#	smartstart:	0
##[BOOT]#	vumeter:	false
##[BOOT]#	softapdelay:	0
##[BOOT]#	flipscreen:	true
##[BOOT]#	invertdisplay:	false
##[BOOT]#	showweather:	true
##[BOOT]#	buttons:	left=16, center=25, right=17, up=255, down=255, mode=255, pullup=true
##[BOOT]#	encoders:	l1=255, b1=255, r1=255, pullup=true, l2=255, b2=255, r2=255, pullup=true
##[BOOT]#	ir:		15
##[BOOT]#	SD:		13
##[BOOT]#	------------------------------------------------
##[BOOT]#	display.init	done
##[BOOT]#	player.init	done
##[BOOT]#	network.begin
##[BOOT]#	Attempt to connect to Wifi_2GHz
##[BOOT]#	..
##[BOOT]#	done
##[BOOT]#	SD search	SD Mounted
##[BOOT]#	netserver.begin	done
##[BOOT]#	telnet.begin	done
##[BOOT]#
##[BOOT]#	Ready! Go to http:/192.168.1.49/ to configure
##[BOOT]#	------------------------------------------------
##[BOOT]#
##CLI.META#: [gotowy]
##WEATHER###: description: zachmurzenie małe, temp:12.3 C, pressure:756mmHg, humidity:60%
##CLI.META#: [łączenie]
##ERROR#:	Request http:// failed!
##ERROR#:	Request http:// failed!
##CLI.META#: ёRadio
##ERROR#:	Error connecting to 
##CLI.META#: Error connecting to 
##ERROR#:	Error connecting to 
##CLI.META#: Error connecting to 
##CLI.INFO#
##SYS.DATE#: 2023-05-19T21:35:27+03:00
##CLI.NAMESET#: 0 ёRadio
##CLI.VOL#: 47
##CLI.STOPPED#
##CLI.META#: Error connecting to 
##CLI.VOL#: 47
##CLI.INFO#
##SYS.DATE#: 2023-05-19T21:35:27+03:00
##CLI.NAMESET#: 0 ёRadio
##CLI.VOL#: 47
##CLI.STOPPED#
##CLI.META#: [łączenie]
##ERROR#:	Request http:// failed!
##ERROR#:	Request http:// failed!
##CLI.META#: ёRadio
##ERROR#:	Error connecting to 
##CLI.META#: Error connecting to 
##ERROR#:	Error connecting to 
##CLI.META#: Error connecting to 
##CLI.INFO#
##SYS.DATE#: 2023-05-19T21:35:29+03:00
##CLI.NAMESET#: 0 ёRadio
##CLI.VOL#: 47
##CLI.STOPPED#

e2002, 2023-05-19:
@hevet  Wygląda na to, że skończyło się miejsce na SPIFFS, ponieważ lista odtwarzania jest dość duża.
Utwórz ESP sketch data upload z IDE i importuj listę odtwarzania bez podłączania karty SD

hevet, 2023-05-19:
@e2002 nic to nie zmieniło, skróciłem nawet listę do 35 stacji i dalej to samo.

e2002, 2023-05-19:
Najwyraźniej jest to błąd w funkcji importu lista odtwarzania. Próbować:
1. Skopiuj pliki playlist.csv i wifi.csv do katalogu yoRadio/data/data/
2. W Arduino IDE wybierz ESP sketch data upload
Będzie działać?
[image]

e2002, 2023-05-19:
@hevet 
Poprawiłem błąd importu:
Zastąp plik yoRadio/src/core/netserver.cpp
[netserver.cpp.zip](https://github.com/e2002/yoradio/files/11520783/netserver.cpp.zip)

e2002, 2023-05-19:
@hevet 
dostępna wersja 0.9.201

hevet, 2023-05-19:
@e2002 Podmiana pliku pomogła :).

e2002, 2023-05-20:
Fixed in version 0.9.201

closed 2023-05-20 by e2002