e2002/yoradio #41: reboot during startup

djeim, 2023-07-13:
Hi all. I assembled the radio according to the scheme on ESP32WROOM, VS1053 and an ecoder. There are two problems
1. During the start of the radio station, the name of the track is displayed, there is no sound, and after 3-4 seconds the radio restarts. The port monitor shows this:
`##CLI.META#: [connecting]
##AUDIO.INFO#: Connect to new host: "https://gamaun.online:8025/radio"
##AUDIO.INFO#: Song stopped incorrectly!
##AUDIO.INFO#: after song stopped incorrectly
##AUDIO.INFO#: REG            dec      bin               hex
##AUDIO.INFO#: -----------  -------  ----------------  -------
##AUDIO.INFO#: MODE          02056   0000100000001000   0808
##AUDIO.INFO#: STATUS        00128   0000000010000000   0080
##AUDIO.INFO#: BASS          00000   0000000000000000   0000
##AUDIO.INFO#: CLOCKF        00000   0000000000000000   0000
##AUDIO.INFO#: DECODE_TIME   00128   0000000010000000   0080
##AUDIO.INFO#: AUDATA        00000   0000000000000000   0000
##AUDIO.INFO#: WRAM          00000   0000000000000000   0000
##AUDIO.INFO#: WRAMADDR      22528   0101100000000000   5800
##AUDIO.INFO#: HDAT0         00000   0000000000000000   0000
##AUDIO.INFO#: HDAT1         00128   0000000010000000   0080
##AUDIO.INFO#: AIADDR        00128   0000000010000000   0080
##AUDIO.INFO#: VOL           00128   0000000010000000   0080
##AUDIO.INFO#: AICTRL0       00000   0000000000000000   0000
##AUDIO.INFO#: AICTRL1       00000   0000000000000000   0000
##AUDIO.INFO#: AICTRL2       00000   0000000000000000   0000
##AUDIO.INFO#: AICTRL3       00128   0000000010000000   0080
##AUDIO.INFO#: PSRAM not found, inputBufferSize: 11903 bytes
##AUDIO.INFO#: SSL has been established in 8851 ms, free Heap: 140608 bytes
##CLI.INFO#
##SYS.DATE#: 2023-07-13T13:00:24+03:00
##CLI.NAMESET#: 1 1
##CLI.META#: [connecting]
##CLI.VOL#: 120
##CLI.PLAYING#
##AUDIO.INFO#: format is mp3
##AUDIO.BITRATE#: 256000
##AUDIO.INFO#: icy-name: Радио Гамаюн
##AUDIO.INFO#: stream ready
##AUDIO.INFO#: buffer filled in 4 ms
##AUDIO.INFO#: StreamTitle='Сколот - Кручина'
##CLI.META#: Сколот - Кручина
##AUDIO.INFO#: slow stream, dropouts are possible
##AUDIO.INFO#: slow stream, dropouts are possible
##AUDIO.INFO#: slow stream, dropouts are possible
##AUDIO.INFO#: slow stream, dropouts are possible
##AUDIO.INFO#: slow stream, dropouts are possible
##AUDIO.INFO#: slow stream, dropouts are possible
##AUDIO.INFO#: slow stream, dropouts are possible
##AUDIO.INFO#: slow stream, dropouts are possible
##AUDIO.INFO#: slow stream, dropouts are possible
##AUDIO.INFO#: slow stream, dropouts are possible
##AUDIO.INFO#: slow stream, dropouts are possible
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40093b48  PS      : 0x00060e33  A0      : 0x800944be  A1      : 0x3ffcd140  
A2      : 0x3ffcadb4  A3      : 0x3ffdd57c  A4      : 0x1a60a54e  A5      : 0x734a6229  
A6      : 0x00060e20  A7      : 0x00000001  A8      : 0x00000001  A9      : 0x00000003  
A10     : 0x204411cc  A11     : 0x000000c8  A12     : 0x3ffcadd8  A13     : 0x3ffcae08  
A14     : 0x00060e23  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x1a60a55a  LBEG    : 0x4008a5a0  LEND    : 0x4008a5b6  LCOUNT  : 0xffffffff  

Backtrace: 0x40093b45:0x3ffcd140 0x400944bb:0x3ffcd160 0x40094728:0x3ffcd180 0x4008431a:0x3ffcd1a0 0x4008432d:0x3ffcd1d0 0x400840a9:0x3ffcd1f0 0x40193bc1:0x3ffcd210 0x4019434c:0x3ffcd240 0x40165c05:0x3ffcd270 0x40190474:0x3ffcd290 0x40191db1:0x3ffcd2d0

ELF file SHA256: e7ca015860ce4ee2

Rebooting...
`

2. I downloaded with the Russian language, but the language is displayed with symbols.
[image]

e2002, 2023-07-13:
```
##AUDIO.INFO#: Song stopped incorrectly!
##AUDIO.INFO#: after song stopped incorrectly
##AUDIO.INFO#: REG dec bin hex
##AUDIO.INFO#: ----------- ------- ---------------- -------
##AUDIO.INFO#: MODE 02056 0000100000001000 0808
##AUDIO.INFO#: STATUS 00640 0000001010000000 0280
##AUDIO.INFO#: BASS 61967 1111001000001111 F20F
##AUDIO.INFO#: CLOCKF 24576 0110000000000000 6000
##AUDIO.INFO#: DECODE_TIME 00128 0000000010000000 0080
##AUDIO.INFO#: AUDATA 44101 1010110001000101 AC45
##AUDIO.INFO#: WRAM 65535 1111111111111111 FFFF
##AUDIO.INFO#: WRAMADDR 22528 0101100000000000 5800
##AUDIO.INFO#: HDAT0 00000 0000000000000000 0000
##AUDIO.INFO#: HDAT1 00128 0000000010000000 0080
##AUDIO.INFO#: AIADDR 00080 0000000001010000 0050
##AUDIO.INFO#: VOL 05397 0001010100010101 1515
##AUDIO.INFO#: AICTRL0 00000 0000000000000000 0000
##AUDIO.INFO#: AICTRL1 00000 0000000000000000 0000
##AUDIO.INFO#: AICTRL2 00000 0000000000000000 0000
##AUDIO.INFO#: AICTRL3 00128 0000000010000000 0080
```
Assembly errors, poor contact, poor quality power supply. It's not about the code.

_>> I downloaded with the Russian language_
Replace file Adafruit_GFX_Library/glcdfont.c with file yoRadio/fonts/glcdfont.c and restart IDE

djeim, 2023-07-13:
I solved the problem as I was advised on 4pda. Added the line 

> #define VS_PATCH_ENABLE false 

to the configuration file. The reboot is gone. Yes, I changed the language. Thank you!)

closed 2023-07-13 by djeim