e2002/yoradio #43: ESP32-Audio-Kit 2.3 with ES8388 audio chip version

SharipovIR, 2023-08-16:
Хочу использовать ESP32-Audio-Kit 2.3 with ES8388 audio chip version. 
Подключил TFT 320x240 с ST7789.
Не могу настроить звук для ёРадио - нужна помощь.
Вдохновляет рабочий пример с веб-радио для этой платы (ESP32 ES8388) из библиотеки ESP32-audioI2S.

e2002, 2023-08-16:
@SharipovIR  I find it difficult to answer this question, because I do not have this board. I am not able to check if it works.

SharipovIR, 2023-08-21:
It's work ! 
For this board need small changes yoRadio.ino , myoptions.h (files in 
[yoRadio.zip](https://github.com/e2002/yoradio/files/12396253/yoRadio.zip) ) .

[image]

brezel64, 2023-11-18:
Hello,
ESP32 Audio Kit 2.2 A247 Board after compiling
Serial monitor output: Connect to ES8388 codec... Failed!
I have the following PIN configuration in the ino:

// GPIOs for SPI ES8388
#define ES_SPI_MOSI 15
#define ES_SPI_MISO 2
#define ES_SPI_SCK 14
#define ES_SD_CS 13

// I2C GPIOs for ES8388
#define ES_IIC_CLK 32
#define ES_IIC_DATA 33

I have the following PIN configuration in myoptions.h:

#define DSP_MODEL DSP_SSD1306
#define I2C_SDA 21
#define I2C_SCL 22
#define I2C_RST -1

#define I2S_DOUT 26
#define I2S_BCLK 27
#define I2S_LRC 25
#define I2S_DSIN 35
#define I2S_MCLK 0

#define BTN_LEFT 5
#define BTN_CENTER 23
#define BTN_RIGHT 18
#define BTN_MODE 19

The audio board runs with the schreiblaul library in my sketch.

Can anyone help here?

SharipovIR, 2024-03-29:
I only succeeded with the SPI display.

Szewcson, 2024-11-24:
@e2002 will it be possible to add driver proposed by @SharipovIR to the project? This codec is also used on esp32-LyraT board. Or maybe it will be worth to integrate [arduino-audio-driver](https://github.com/pschatzmann/arduino-audio-driver) since it also support es8311 used by esp32-LyraT-Mini.

andrew-jar, 2024-12-29:
Firmware Yoradio v0.9.412

File myoptions.h:

```
#define L10N_LANGUAGE      RU

#define DSP_MODEL			DSP_ILI9341
#define TFT_CS            5                 /*  SPI CS pin  */
#define TFT_RST          -1               /*  SPI RST pin.  set to -1 and connect to Esp EN pin */
#define TFT_DC           15 

//MOSI: 23
//MISO: 19
//SCK: 18
//SS: 5

//Korekta: I2C wynosi 23/18 lub 32/33 w zależności od wariantu płyty
//Interfejs sprzętowy jest inny, IIC V2.3 to 32, 33, a IIC V2.2 18, 23

#define I2S_DOUT      26
#define I2S_BCLK      27
#define I2S_LRC       25
#define I2S_DSIN      35
#define I2S_MCLK      0

#define ENC2_BTNR			22  //CLK
#define ENC2_BTNL			14  //DT 
#define ENC2_BTNB			13  //SW 
#define ENC2_INTERNALPULLUP			true

#define IR_PIN	12

#endif
```
[image]
main.cpp file to replace
[main.cpp_for_ESP32_A1S_ES8388.zip](https://github.com/user-attachments/files/18270333/main.cpp_for_ESP32_A1S_ES8388.zip)

Thoky81, 2025-04-05:
Hello, is there any chance to connect I2C display?

dgel27, 2025-06-11:
> Hello, is there any chance to connect I2C display?

Sinse ES8388 used I2C bus for it control, and there his SDA and ICL come out of the module, you can connect to it other devices, it only 2 ways use it:
1. Init ES8388 and release wire for other devices. This is less preferable option, because you can not control ES8388 during operation (still not implemented here, but it have unlimited possibilities for suound quality and switch/mix devices)
2. Use TwoWire class. With this option possible create 2 I2C undepended buses