e2002/yoradio #44: Encoder functioanlity

ESPNoodle, 2023-09-07:
Just a cheeky request....   

so when using the encoder i have noticed the Station list moves up when I turn the encoder clockwise, the opposite is true for the volume (it goes down on clockwise). when in my head... 😶 clockwise should be up in volume.

Again, can't thank you enough for this awesome project, managed to fit a oled radio in an audio cassette tape!

e2002, 2023-09-07:
@ESPNoodle 
Swap ENC_BTNL and ENC_BTNR

ESPNoodle, 2023-09-08:
Hello (Sir?) 

I fully understand swapping od ENC_BTNR -L. 
this is not what I am getting at.

In encoder 1 or 2 modes (either it does the same). The clockwise for volume increases it. the clockwise on station list decreases it.
(I too am aware that if the station numbers are on it makes logical sense).... But! with numbers off it just feels opposite to the volume action (of up/down).

Is there a way I can edit the code just for my own silly needs... LOL!!! 😂😂😂

 is there a way I can pay (donate) for your amazing work too? 
and thank you for putting up with me. 

Best Regards.