e2002/yoradio #49: Arduino IDE compile time very long lately

dabu667, 2023-09-20:
Hi,

I am using IDE 1.8.19 (but also tried 1.8.13) and a few versions of ESP core and I have the same problem: The compilation time takes over 12 minutes (i5 processor 16GB RAM machine, win 10). I remember that when I first found this project the compile time was under 2 minutes. I know this is not because of your code but I was wondering is anybody else has the same problem.
I'm guessing it has to do something with the esp core version and the compiler: enabling verbose output it shows some very long (over 8000 chars) command lines.

Dan

SharipovIR, 2023-09-27:
Disabling the antivirus helped me.

dabu667, 2023-09-27:
@SharipovIR thank you for trying to help. Unfortunately I already tried that but with no effect. For example, if I enable the IR option (the library has a lot of files) the first compile takes over 12 minutes. I tried several versions of IDE, ESP core, I even tried to change the version of the cgw compiler. Even for the following compiles the time it takes is similar: I guess it recompiles every library again.
Testing different solutions is a very time consuming matter, so I gave up hoping to find an answer here.
Anyway, this is not an issue with yoRadio so if anyone thinks this is off topic I will close it.