e2002/yoradio #50: ESP32 core 2.0.13 not working with this project

dabu667, 2023-09-20:
I updated esp32 core board in Arduino IDE to the latest version: trying to play a station results in resetting the board:

> ##CLI.META#: [connecting]
> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
> 
> Core  1 register dump:
> PC      : 0x400ff513  PS      : 0x00060a30  A0      : 0x800ff61c  A1      : 0x3ffb1a30  
> A2      : 0x00000000  A3      : 0x3f4001e0  A4      : 0x0000002d  A5      : 0x3ffb18c0  
> A6      : 0x0000002d  A7      : 0x3ffb18c0  A8      : 0x00000000  A9      : 0x00000042  
> A10     : 0x00000069  A11     : 0x00000075  A12     : 0x00000006  A13     : 0x00001000  
> A14     : 0x00000006  A15     : 0x3ffb18e4  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
> EXCVADDR: 0x00000010  LBEG    : 0x4008bf6d  LEND    : 0x4008bf7d  LCOUNT  : 0xfffffffe  
> 
> Backtrace: 0x400ff510:0x3ffb1a30 0x400ff619:0x3ffb1a60 0x400e498a:0x3ffb1a80 0x400e4d36:0x3ffb1d20 0x400f8fdd:0x3ffb1fc0 0x400f9105:0x3ffb2240 0x400d29a0:0x3ffb2270 0x4011439d:0x3ffb2290
> 
> ELF file SHA256: e6c31fe8c17b69b9
> 
> Rebooting...
> ets Jul 29 2019 12:21:46
> 
> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0030,len:1344
> load:0x40078000,len:13964
> load:0x40080400,len:3600
> entry 0x400805f0
> ##[BOOT]#	************************************************
> ##[BOOT]#	*               ёPadio v0.9.260                *
> ##[BOOT]#	************************************************
> ##[BOOT]#	------------------------------------------------
> ##[BOOT]#	arduino:	10813
> ##[BOOT]#	compiler:	8.4.0
> ##[BOOT]#	esp32core:	2.0.13
> ##[BOOT]#	chip:		model: ESP32-D0WD-V3 | rev: 3 | id: 16396048 | cores: 2 | psram: 0
> ##[BOOT]#	display:	9
> ##[BOOT]#	audio:		I2S (27, 26, 25)
> ##[BOOT]#	audioinfo:	false
> ##[BOOT]#	smartstart:	0
> ##[BOOT]#	vumeter:	false
> ##[BOOT]#	softapdelay:	0
> ##[BOOT]#	flipscreen:	false
> ##[BOOT]#	invertdisplay:	false
> ##[BOOT]#	showweather:	false
> ##[BOOT]#	buttons:	left=255, center=255, right=255, up=255, down=255, mode=255, pullup=true
> ##[BOOT]#	encoders:	l1=255, b1=255, r1=255, pullup=true, l2=255, b2=255, r2=255, pullup=true
> ##[BOOT]#	ir:		27
> ##[BOOT]#	SD:		5
> ##[BOOT]#	------------------------------------------------
> ##[BOOT]#	SPIFFS mounted
> ##[BOOT]#	display.init	done
> ##[BOOT]#	player.init	done
> ##[BOOT]#	network.begin
> ##[BOOT]#	Attempt to connect to mxrp_guest
> ##[BOOT]#	...
> ##[BOOT]#	done
> ##[BOOT]#	SD search	SD Mount Failed
> ##[BOOT]#	netserver.begin	done
> ##[BOOT]#	telnet.begin	done
> ##[BOOT]#
> ##[BOOT]#	Ready! Go to http:/192.168.0.163/ to configure
> ##[BOOT]#	------------------------------------------------
> ##[BOOT]#
> ##CLI.META#: [ready]
> Lost connection, reconnecting...
> Lost connection, reconnecting...
> Lost connection, reconnecting...

Also tried 2.0.12 - not working, 2.0.10 works.
Have a picture with my latest build:
[image]

e2002, 2023-09-22:
Hi @dabu667 
Fixed in version [0.9.273](https://github.com/e2002/yoradio#v09273)

dabu667, 2023-09-22:
I can confirm that v0.9.273 works with esp32 core 2.0.13. Thank you!
There are various clones of ESP32devKit from the Chinese market and some of them have weak antenna design or something and the Wi-Fi signal is weak. I was wondering if there a way to boost the signal a bit from software. Will some of this commands do it?
```
WiFi.setTxPower(WIFI_POWER_19_5dBm);    // Set Wi-Fi RF power output to highest level
WiFi.setOutputPower(20.5);    // Sets Wi-Fi RF power output to highest level, highest RF power usage
esp_wifi_set_ps(WIFI_PS_NONE);   //disable modem sleep entirely
```

If yes, where to put the code?

e2002, 2023-09-23:
@dabu667 
No, these functions will not increase the received signal level. If the signal quality is poor, you can try using an external antenna.

dabu667, 2023-09-25:
Thank you. I was hoping there was an easy fix.

closed 2023-09-25 by dabu667