e2002/yoradio #55: When I turn on any station, the VU meter indicators are always at the maximum level.

hevet, 2023-10-19:
Hi. I used your PCB design https://github.com/e2002/yopcb/tree/main/vs1053shield_v2, the first board was assembled, the radio plays nicely. I am using the display on ILI9341. There is one problem, however, when I turn on any station, the VU meter indicators are always at the maximum reading, they do not change, when I switch the station they turn off and after starting the station they are at the maximum reading again. Is there any way to solve this problem?

e2002, 2023-10-19:
@hevet Hello! There might be a poor soldering joint or a short circuit between the pins of the VS1053. Pay special attention to the MISO pin (pin 30). You could also try replacing the ESP32 devkit. In short, the issue is somewhere in MISO, where it's capturing the sound level readings.

hevet, 2023-10-19:
Thanks for the answer. I checked the soldering, there are no shorts and the soldering looks good. Tomorrow I will carefully check the soldering under a magnifying glass.
I also have a question: can VS1003 be used instead of the VS1053 codec and will it work properly without any modifications? I know that VS1003 does not support AAC.

Ps. Is it possible for you to share the entire board design with Kicad, not just the Gerber files?

e2002, 2023-10-20:
@hevet On VS1003, the VUmeter won't work; it's not supported hardware-wise

hevet, 2023-10-20:
@e2002 So you found the cause, it turns out that the VS1003B codec is the cause, VS1053b should arrive today, I will replace it and see if it works

closed 2023-10-21 by hevet