e2002/yoradio #57: VU meter sensitivity

stjames12, 2023-11-14:
Hello, I have a small problems with the VU meter functionality, I am using VS1053 board to connect via ESP32, but the VU meter does not working, it is shoe me always the maximum level, somebody can help me to solve this mistake?
Best regards, IMI!

e2002, 2023-11-14:
Hi @stjames12 Please send a photo of the VS1053 module. And one more photo where the inscription on the chip is visible.

stjames12, 2023-11-14:
Hi, there are (it's looks like the chip is VS1003)
[image]
[image]

e2002, 2023-11-14:
@stjames12 The VUmeter does not work on VS1003 chips; this is a hardware limitation.

stjames12, 2023-11-14:
> @stjames12 The VUmeter does not work on VS1003 chips; this is a hardware limitation.

Ok, thank's a lot!

acoveca, 2023-11-17:
I have VS1053 and the VU-meter doesn't work either. Just during station change, it comes very short and disappear.

[image]
[image]

e2002, 2023-11-17:
@acoveca No contact between VS1053 MISO and GPIO19

acoveca, 2023-11-17:
I connected the VS1053 (and included SD ) module to HSPI (Pins 12, 13, 14) instead of VSPI (Pins 18,19 and 23), so is VS1053 MISO  is connected to pin 12 and not to 19. The complete connections is attached.
[YoRadio_Pinout_20231117.pdf](https://github.com/e2002/yoradio/files/13396732/YoRadio_Pinout_20231117.pdf)

Willie131129, 2023-11-24:
@e2002 
I have the same problems with the VS1053 Board and the VUmeter not working. I have checked triple times the connections
and the meter is still not working, only showing up for a short glitch after switching it on. i used HW-SPI interface for the VS1053
 so PIN G25=DCS,G26=DREQ,G27=CS,G14=SCK,G12=MISO,G13=MOSI and EN=RST are used (suggested by config website).
I am using the "green Board" 
https://www.amazon.de/dp/B07DJZ8HCL?psc=1&ref=ppx_yo2ov_dt_b_product_details