e2002/yoradio #59: Question about compiling firmware in PlatformIO

Victuar98, 2023-12-13:
Dear community, how to compile firmware for YoRado in the Platformio environment, with a single bin file? Or how to update the firmware from three files via W-Fi? Thanks to everyone who responded!)

DonEugraf, 2024-02-03:
unfortunately its not made for platformio :(
you have to use arduino ide 1.9.x not even 2.x

bad040, 2024-02-13:
I have converted all files to platformio. Can easily be done

DonEugraf, 2024-02-24:
would you mind sharing it please?

Victuar98, 2024-02-28:
The firmware compiles faster in the Platform than in the Arduino IDE, but only the monolithic bin file is updated over Wi-Fi, and I had to put up with it...

Victuar98, 2024-03-08:
platformio.ini

[env:esp-wrover-kit]
platform = espressif32
board = esp-wrover-kit //(or your model board)
board_build.mcu = esp32
board_build.f_cpu = 240000000L //(according to specification)
framework = arduino
monitor_speed = 115200
upload_port = COM4 //(your nomber port)
upload_protocol = esptool
lib_extra_dirs = ${workspacedir}
// required libraries
lib_deps = 
	crankyoldgit/IRremoteESP8266 @ ^2.8.6
	adafruit/Adafruit GFX Library@^1.11.9
	adafruit/Adafruit BusIO@^1.14.5

trip5, 2024-12-27:
I'll share my platformio.ini file as well.  I use my fork directly so the entire folder structure is unchanged... and I do use an ESP32-S3... so you may need to edit some of that.
```
[platformio]
src_dir = ./yoRadio/src
data_dir = ./yoRadio/data

[env:esp32-s3-n16r8]
lib_extra_dirs = ./
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino

upload_port = COM15
monitor_port = COM15

monitor_speed = 115200
build_flags =
	-DCORE_DEBUG_LEVEL=5	; Set to 0 for no debug; saves memory
                                ; Set to 5 for full debug

; required libraries - you may need to edit this list!
lib_deps =
  crankyoldgit/IRremoteESP8266 @ ^2.8.6
  adafruit/Adafruit GFX Library@^1.11.9
  adafruit/Adafruit BusIO@^1.14.5
  adafruit/Adafruit SSD1306@^2.5.13
  adafruit/Adafruit SH110X@^2.1.11
  ;ESPmDNS

; ESP32-S3 info from: https://github.com/sivar2311/ESP32-S3-PlatformIO-Flash-and-PSRAM-configurations
board_build.arduino.memory_type = qio_opi
board_build.flash_mode = qio
board_build.psram_type = opi
board_upload.flash_size = 16MB
board_upload.maximum_size = 16777216
board_build.extra_flags = 
  -DBOARD_HAS_PSRAM