e2002/yoradio #60: ESP32 S3

ESPNoodle, 2023-12-13:
Hello Sir!

Just wondering if an option to have ESP32 S3 chip could be added to "myoptions-generator.html" could possibly be added.
the S3 has onboard USB-OTG, so it allows for us as makers to use just the ESP module (no UART) to still connect directly on very slim hardware. 
What stops me is the pin out and GPIO numbers are all different.
I know it's probably a big ask, maybe a fellow wizard could help?

Big thanks again for such an awesome project!

trip5, 2024-12-29:
Sorry, I kind of duplicated your issue but, I did successfully get it working on ESP32-S3: https://github.com/e2002/yoradio/issues/123

trip5, 2025-01-05:
Wanna take a look? https://trip5.github.io/yoradio-docs/docs/myoptions-generator.html

trip5, 2025-01-05:
By the way if you have a particular S3 board you use, tell me (show me) and I can add it.