e2002/yoradio #65: ILI9341 display can't rotate

fenyvesi, 2023-12-22:
I can't rotate the display in the proper position. I tried:
1. #define TFT_ROTATE 3 // 3=270 degrees, 
2. #define ROTATE_90   true             /*  Optional 90 degree rotation for square displays */

The myoptions.h file:
```

#ifndef myoptions_h
#define myoptions_h
#define DSP_MODEL DSP_ILI9341
#define DSP_HSPI true
#define LED_INVERT true
#define TFT_CS 15
#define TFT_DC 2
#define TFT_RST -1
#define LED_BUILTIN 17
#define BRIGHTNESS_PIN 21
#define I2S_INTERNAL true
#define PLAYER_FORCE_MONO true
#define TS_MODEL TS_MODEL_XPT2046
#define TS_SPIPINS 25, 39, 32 /* SCK, MISO, MOSI */
#define TS_CS 33
#define TFT_ROTATE 1 // 3=270 degrees, nem megy
//#define ROTATE_90   true             /*  Optional 90 degree rotation for square displays */
#define LIGHT_SENSOR 34
#define AUTOBACKLIGHT_MAX 1024
#endif
```
Thank you in advance.

George

e2002, 2023-12-22:
Rectangular displays can only be rotated 180 degrees. This is done in the web interface.

[image]

fenyvesi, 2023-12-22:
Yes I saw it but I need 90/270 degrees.

e2002, 2023-12-22:
Portrait orientation is not currently supported.

fenyvesi, 2023-12-26:
The problem arised in using of the "Cheap Yellow Display". I suggest that the problem lies in the fact that this is not a 320x240 but a 240x320 display and somehow I should change the orientation with the max line and column number.

fenyvesi, 2023-12-26:
I just realized thet the display is ST7789 and everything is fine. Thank you.

closed 2023-12-26 by fenyvesi