e2002/yoradio #67: yoRadio on ESP32-8048s070C?

echosmart792, 2023-12-25:
Can this project also work on ESP32-8048s070C? It has an ESP32-S3 microcontroller, a 7-inch capacitive touchscreen display, and a built-in i2C audio amplifier.

echosmart792, 2024-02-22:
I can't find any link where I can find the necessary documentation. Can you help me with a link?

dabu667, 2024-02-23:
Seems unlikely to work on ESP32-8048s070C. The chipset is ESP32-S3 and this project is meant for the original ESP32. Also, I see the TFT driver is EK9716, which is not included in this project, and probably the big resolution takes too much memory. I may be wrong though, and maybe someone managed to make it work. I'm just trying to help, as others have helped me.