e2002/yoradio #71: esp32-2432s028 with PSRAM

dabu667, 2024-01-24:
HI,
I removed the RGB LED to free IO16 and IO17 from the board and attached PSRAM IC following this tutorial:
[https://github.com/hexeguitar/ESP32_TFT_PIO?tab=readme-ov-file#3-adding-psram](url)
Everything seems fine until I try to play a stream - see serial output below:

> ets Jul 29 2019 12:21:46
> 
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0030,len:1344
> load:0x40078000,len:13964
> load:0x40080400,len:3600
> entry 0x400805f0
> [   459][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
> [   459][I][Audio.cpp:198] Audio(): internal DAC
> [   474][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
> ##[BOOT]#	************************************************
> ##[BOOT]#	*               ёPadio v0.9.300                *
> ##[BOOT]#	************************************************
> ##[BOOT]#	------------------------------------------------
> ##[BOOT]#	arduino:	10819
> ##[BOOT]#	compiler:	8.4.0
> ##[BOOT]#	esp32core:	2.0.14
> ##[BOOT]#	chip:		model: ESP32-D0WD-V3 | rev: 3 | id: 16396048 | cores: 2 | psram: 4192123
> ##[BOOT]#	display:	9
> ##[BOOT]#	audio:		I2S (27, 26, 25)
> ##[BOOT]#	audioinfo:	false
> ##[BOOT]#	smartstart:	0
> ##[BOOT]#	vumeter:	false
> ##[BOOT]#	softapdelay:	0
> ##[BOOT]#	flipscreen:	false
> ##[BOOT]#	invertdisplay:	false
> ##[BOOT]#	showweather:	false
> ##[BOOT]#	buttons:	left=255, center=255, right=255, up=255, down=255, mode=255, pullup=true
> ##[BOOT]#	encoders:	l1=255, b1=255, r1=255, pullup=true, l2=255, b2=255, r2=255, pullup=true
> ##[BOOT]#	ir:		255
> ##[BOOT]#	SD:		5
> ##[BOOT]#	------------------------------------------------
> ##[BOOT]#	SPIFFS mounted
> ##[BOOT]#	display.init	done
> ##[BOOT]#	player.init	done
> ##[BOOT]#	network.begin
> [  1141][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
> ##[BOOT]#	Attempt to connect to dabu
> [  1213][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
> ##[BOOT]#	.[  1324][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
> [  1324][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 202 - AUTH_FAIL
> [  1327][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi Reconnect Running
> [  1391][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
> [  1418][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
> [  1418][D][WiFiGeneric.cpp:1102] _eventCallback(): STA IP: 192.168.1.100, MASK: 255.255.255.0, GW: 192.168.1.1
> .
> ##[BOOT]#	done
> ##[BOOT]#	SD search	[  1728][W][sd_diskio.cpp:174] sdCommand(): no token received
> [  1828][W][sd_diskio.cpp:174] sdCommand(): no token received
> [  1928][W][sd_diskio.cpp:174] sdCommand(): no token received
> [  2028][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
> [  2028][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
> [  2029][E][sd_diskio.cpp:805] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
> [  2038][W][sd_diskio.cpp:174] sdCommand(): no token received
> [  2143][W][sd_diskio.cpp:174] sdCommand(): no token received
> [  2243][W][sd_diskio.cpp:174] sdCommand(): no token received
> [  2343][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
> SD Mount Failed
> ##[BOOT]#	netserver.begin	done
> ##[BOOT]#	telnet.begin	done
> ##[BOOT]#
> ##[BOOT]#	Ready! Go to http:/192.168.1.100/ to configure
> ##[BOOT]#	------------------------------------------------
> ##[BOOT]#
> ##CLI.META#: [ready]
> [WEBSOCKET] client #1 connected from 192.168.1.171
> ##CLI.META#: [ready]
> ##CLI.VOL#: 12
> ##CLI.INFO#
> ##SYS.DATE#: 2024-01-24T11:39:07+03:00
> ##CLI.NAMESET#: 1 Paradise
> ##CLI.VOL#: 12
> ##CLI.STOPPED#
> [WEBSOCKET] client #1 disconnected
> [WEBSOCKET] client #2 connected from 192.168.1.171
> [WEBSOCKET] client #2 disconnected
> [WEBSOCKET] client #3 connected from 192.168.1.171
> ##CLI.META#: [ready]
> ##CLI.VOL#: 12
> ##CLI.INFO#

.. but when I hit play I get constant Wi-Fi disconnect and reconnect:

> ##SYS.DATE#: 2024-01-24T11:39:29+03:00
> ##CLI.NAMESET#: 1 Paradise
> ##CLI.VOL#: 12
> ##CLI.STOPPED#
> ##CLI.META#: [connecting]
> ##AUDIO.INFO#: Connect to new host: "http://stream.radioparadise.com/aac-128"
> ##AUDIO.INFO#: PSRAM found, inputBufferSize: 283615 bytes
> ##AUDIO.INFO#: buffers freed, free Heap: 153056 bytes
> ##AUDIO.INFO#: Connection has been established in 100 ms, free Heap: 152204 bytes
> ##CLI.INFO#
> ##SYS.DATE#: 2024-01-24T11:39:51+03:00
> ##CLI.NAMESET#: 1 Paradise
> ##CLI.META#: [connecting]
> ##CLI.VOL#: 12
> ##CLI.PLAYING#
> ##AUDIO.INFO#: format is aac
> ##AUDIO.BITRATE#: 128000
> ##AUDIO.INFO#: icy-name: Radio Paradise (128k aac)
> ##AUDIO.INFO#: AACDecoder has been initialized, free Heap: 130548 bytes
> [ 58823][W][AsyncTCP.cpp:965] _poll(): ack timeout 4
> [WEBSOCKET] client #3 disconnected
> [ 61963][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
> [ 61963][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 200 - BEACON_TIMEOUT
> [ 61967][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
> Lost connection, reconnecting to dabu...
> [ 62081][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
> [ 62156][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
> [ 62157][D][WiFiGeneric.cpp:1102] _eventCallback(): STA IP: 192.168.1.100, MASK: 255.255.255.0, GW: 192.168.1.1
> ##AUDIO.INFO#: buffers freed, free Heap: 155840 bytes
> ##CLI.META#: [stopped]
> ##CLI.INFO#
> ##SYS.DATE#: 2024-01-24T11:40:01+03:00
> ##CLI.NAMESET#: 1 Paradise
> ##CLI.VOL#: 12
> ##CLI.STOPPED#
> ##CLI.META#: [connecting]
> ##AUDIO.INFO#: Connect to new host: "http://stream.radioparadise.com/aac-128"
> ##AUDIO.INFO#: buffers freed, free Heap: 155616 bytes
> ##AUDIO.INFO#: Connection has been established in 48 ms, free Heap: 154648 bytes
> ##CLI.INFO#
> ##SYS.DATE#: 2024-01-24T11:40:01+03:00
> ##CLI.NAMESET#: 1 Paradise
> ##CLI.META#: [connecting]
> ##CLI.VOL#: 12
> ##CLI.PLAYING#
> ##AUDIO.INFO#: format is aac
> ##AUDIO.BITRATE#: 128000
> ##AUDIO.INFO#: icy-name: Radio Paradise (128k aac)
> ##AUDIO.INFO#: AACDecoder has been initialized, free Heap: 132792 bytes

I tried changing XTASK_MEM_SIZE from 4096 to 8192 to 16384 but nothing changed
I suspect it's a memory issue though: if I use DSP_DUMMY it works fine. Or maybe a SPI issue? Although display and PSRAM use different SPI interfaces
Can you please help?

dabu667, 2024-01-24:
.. to add more info:
I tried setting flash freq to 40 or 80 MHz, setting QIO, DIO, DOUT mode to no avail ..
but I noticed something: the problem only occurs after a software reset:

> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

on the first hardware reset everything works fine:

> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

dabu667, 2024-01-25:
Another day lost trying to find a solution. I removed the PSRAM chip from the board bringing it to the original state and the problem persists: after the first soft reset (from the serial monitor, from the web interface or when waking from deep sleep) the board becomes unstable: wi-fi drops and reconnects randomly, web interface is inaccessible or barely loads, vu-meters hang and I even notice dropouts on another yoRadio device nearby!! so something really wrong happens to the wi-fi chip.
I cannot isolate the problem but I'm pretty sure it is this hardware and software combination:
esp32-2432s028 board - the yellow one documented here:
[https://github.com/e2002/yoradio/wiki/ESP32-TFT-Boards-configurations#esp32-2432s028r](url)
I'm using the IDE 1.8.19, latest esp core 2.0.14, and latest libraries (although I tried many older ones also)
If anybody has this board and is willing to try to flash the latest version (v0.9.300) please leave me a message.

saarfranzose, 2024-01-25:
works fine here with this myoptions (and PCM5102 DAC):

[image]

[image]

only the TP has to rotate 180°

dabu667, 2024-01-26:
Thank you [saarfranzose](https://github.com/saarfranzose) for replying. Unfortunately it doesn't work - it has the same behavior: after the first software restart it constantly drops Wi-Fi. 

Would you be so kind to send me your settings:

Arduino IDE version:
Copmiler version:
ESP32 core version:
All of the above show on serial after reset:

> ##[BOOT]#	arduino:	10813
> ##[BOOT]#	compiler:	8.4.0
> ##[BOOT]#	esp32core:	2.0.10

Dependencies versions: 
Adafruit_GFX:
Adafruit_ILI9341:
OneButton:
XPT2046_Touchscreen:
RTCLib:
If you're using the latest then don't bother.

Flash settings: 40 / 80Mhz, QIO, DIO, QOUT, DOUT?

Can you attach the exported compiled binary (yoRadio.ino.esp32.bin)?

saarfranzose, 2024-01-26:
hi dabu,
my settings are absolutly standard:
[image]
core is 2.0.14
and the libraries, if not included in yoradio/src, have current version

saarfranzose, 2024-01-26:
Hello dabu,

 

here is my bin-file

 

regards

Franz-Josef

dabu667, 2024-01-26:
I don't think you can send a file by replying from email, I think you need to log into github.
Thank you for your help, though.
I have one more thing to ask of you if it's not too difficult, then I give up: do a software reset (boot command on serial or save & reboot on WiFi settings on web interface) and then see if everything is as responsive as before: check for dropouts on stream, vu-meters freezing or unresponsive web interface.

dabu667, 2024-02-01:
It's FIXED now! It was a faulty ESP32 module on this CYD board. 
I have ordered an original esp32 module manufactured by Espressif and replace the one on the board and everything works jus as it should. Is it worth all the time spent? I guess not but I just couldn't let it go.

If you want to find out more read below:
I have tried everything before: in software disconnecting WiFi and disabling various esp peripherals (spi, wifi radio, etc) before restarting but nothing worked. 
Then I modified the hardware:
 - checking / replacing and adding quality capacitors
 - powering the board from a good 3.3V power supply directly on the esp pins
 - disabling / removing CH340 serial converter (I found people talking on forums about faulty CH340 that inject 5V into esp module making it unstable; this was not my case.
 
So, the final step I took was to order a new original Espressif module and replace it. I went for a module with an antenna connector so to exclude the low signal from the possible problems but even if I don't connect the antenna (at -80dB) I have a more responsive device then any of my previous builds using non Espressif modules!
So, be aware! it seems that some esp32 modules are really low quality!
[image]

closed 2024-02-01 by dabu667