e2002/yoradio #73: Kein Bild auf ESP32-2432S028 (CYD)

fetzer91, 2024-01-28:
Hallo,
ich habe alles so wie beschrieben eingerichtet und die myoptions [https://pastebin.com/Fjam0cnC](url) verwendet. Ich konnte im Browser alles einrichten und auch eine Stationsliste hochladen. Aber das Display bleibt schwarz. Was kann passiert sein?

VG Stefan

BeryBurnout, 2024-01-30:
hast die richtigen Einstellungen reinkopiert und die anderen deaktiviert ? hat bei mir mit dem größeren CYD einwandfrei geklappt

fetzer91, 2024-01-30:
> @fetzer91 Es sieht so aus, als ob die Hintergrundbeleuchtung des Displays nicht funktioniert.

Hallo, das CYD war neu und die Demo, welche von Anfang drauf war hat funktioniert. Daher denke ich dass das CYD soweit intakt ist.

fetzer91, 2024-01-30:
> hast die richtigen Einstellungen reinkopiert und die anderen deaktiviert ? hat bei mir mit dem größeren CYD einwandfrei geklappt

Hallo, welche Einstellungen genau meinst du? ich habe die myoptions aus dem Link oben verwendet und sonst nach Anleitung gearbeitet.

e2002, 2024-01-30:
@fetzer91 
Welche Einstellungen haben Sie auf dieser Registerkarte?

[image]

e2002, 2024-01-30:
@fetzer91 
Bitte posten Sie das Startprotokoll vom Serieller Monitor

[image]

fetzer91, 2024-01-30:
> @fetzer91 Welche Einstellungen haben Sie auf dieser Registerkarte?
> 
> ![screen](https://private-user-images.githubusercontent.com/4333855/300913322-f7f71378-2cdd-49dd-8d34-ea01c786e5ab.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY2Mzc1NTksIm5iZiI6MTcwNjYzNzI1OSwicGF0aCI6Ii80MzMzODU1LzMwMDkxMzMyMi1mN2Y3MTM3OC0yY2RkLTQ5ZGQtOGQzNC1lYTAxYzc4NmU1YWIuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDEzMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMzBUMTc1NDE5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZGIyNDQ5MWJhYjNlNTY5N2RjYTMxYmZmZTkyNmNkMmQzMjk4OTYxZTBjMDAxMzJjZDM0ZGIyYzJiZmY5ODg4MCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.SKiSaSvlPuf7uFxr5Z7duwqk7sf4aolae8Ta8LPVhkU)

Ich habe weder screen noch weather in den settings stehen.

e2002, 2024-01-30:
@fetzer91 
> 
> Ich habe weder screen noch weather in den settings stehen.

Wo befindet sich Ihre myoptions.h-Datei?

>
> [screen]

Das angehängte Bild ist nicht sichtbar.

[image]