e2002/yoradio #75: 3/4 of the screen works, other 1/4 doesn't work

TheMiro20, 2024-02-05:
```
Hello every one, i am having the problem that 1/4 of my screen portion doesnt work. I am new to this sort of programming and i have no clue how to solve it.... I building a DIY speaker and I am trying to add this CYD screen with esp internet radio, but no luck with setting up the screen...
![IMG_1113](https://github.com/e2002/yoradio/assets/139749104/0b680ea1-190c-413e-915e-bdf76d126c7c)
Here is a picture how it looks like.

Here is a second picture me trying to solve it but no luck...
![IMG_1112](https://github.com/e2002/yoradio/assets/139749104/de47fccb-5a47-482e-a6be-b0e4ff6cdd0d)

This the IlI9341conf.h code:
/*************************************************************************************
    ILI9341 320x240 displays configuration file.
    Copy this file to yoRadio/src/displays/conf/displayILI9341conf_custom.h
    and modify it
    More info on https://github.com/e2002/yoradio/wiki/Widgets#widgets-description
*************************************************************************************/

#ifndef displayILI9341conf_h
#define displayILI9341conf_h

#define DSP_WIDTH       240
#define DSP_HEIGHT      320  // Swap width and height for landscape mode
#define TFT_FRAMEWDT    8
#define MAX_WIDTH       DSP_WIDTH-TFT_FRAMEWDT*2

#if BITRATE_FULL
  #define TITLE_FIX 44
#else
  #define TITLE_FIX 0
#endif
#define bootLogoTop     68

/* SROLLS  */                            /* {{ left, top, fontsize, align }, buffsize, uppercase, width, scrolldelay, scrolldelta, scrolltime } */
const ScrollConfig metaConf       PROGMEM = {{ TFT_FRAMEWDT, TFT_FRAMEWDT, 3, WA_LEFT }, 140, true, MAX_WIDTH, 5000, 5, 30 };
const ScrollConfig title1Conf     PROGMEM = {{ TFT_FRAMEWDT, 50, 2, WA_LEFT }, 140, true, MAX_WIDTH-TITLE_FIX, 5000, 4, 30 };
const ScrollConfig title2Conf     PROGMEM = {{ TFT_FRAMEWDT, 70, 2, WA_LEFT }, 140, true, MAX_WIDTH-TITLE_FIX, 5000, 4, 30 };
const ScrollConfig playlistConf   PROGMEM = {{ TFT_FRAMEWDT, 112, 2, WA_LEFT }, 140, true, MAX_WIDTH, 1000, 4, 30 };
const ScrollConfig apTitleConf    PROGMEM = {{ TFT_FRAMEWDT, TFT_FRAMEWDT, 3, WA_CENTER }, 140, false, MAX_WIDTH, 0, 4, 20 };
const ScrollConfig apSettConf     PROGMEM = {{ TFT_FRAMEWDT, 240-TFT_FRAMEWDT-16, 2, WA_LEFT }, 140, false, MAX_WIDTH, 0, 4, 30 };
const ScrollConfig weatherConf    PROGMEM = {{ 8, 87, 2, WA_LEFT }, 140, true, MAX_WIDTH, 0, 4, 30 };

/* BACKGROUNDS  */                       /* {{ left, top, fontsize, align }, width, height, outlined } */
const FillConfig   metaBGConf     PROGMEM = {{ 0, 0, 0, WA_LEFT }, DSP_HEIGHT, 38, false };  // Swap width and height
const FillConfig   metaBGConfInv  PROGMEM = {{ 0, 38, 0, WA_LEFT }, DSP_HEIGHT, 1, false };  // Swap width and height
const FillConfig   volbarConf     PROGMEM = {{ TFT_FRAMEWDT, DSP_WIDTH-TFT_FRAMEWDT-6, 0, WA_LEFT }, MAX_WIDTH, 6, true };
const FillConfig  playlBGConf     PROGMEM = {{ 0, 107, 0, WA_LEFT }, DSP_HEIGHT, 24, false };  // Swap width and height
const FillConfig  heapbarConf     PROGMEM = {{ 0, DSP_WIDTH-TFT_FRAMEWDT, 0, WA_LEFT }, DSP_HEIGHT, 1, false };  // Swap width and height

/* WIDGETS  */                           /* { left, top, fontsize, align } */
const WidgetConfig bootstrConf    PROGMEM = { 0, 182, 1, WA_CENTER };
const WidgetConfig bitrateConf    PROGMEM = { 70, 191, 1, WA_LEFT };
const WidgetConfig voltxtConf     PROGMEM = { 0, 214, 1, WA_CENTER };
const WidgetConfig  iptxtConf     PROGMEM = { TFT_FRAMEWDT, 214, 1, WA_LEFT };
const WidgetConfig   rssiConf     PROGMEM = { TFT_FRAMEWDT, 214-6, 2, WA_RIGHT };
const WidgetConfig numConf        PROGMEM = { 0, 120+30, 52, WA_CENTER };
const WidgetConfig apNameConf     PROGMEM = { TFT_FRAMEWDT, 66, 2, WA_CENTER };
const WidgetConfig apName2Conf    PROGMEM = { TFT_FRAMEWDT, 90, 2, WA_CENTER };
const WidgetConfig apPassConf     PROGMEM = { TFT_FRAMEWDT, 130, 2, WA_CENTER };
const WidgetConfig apPass2Conf    PROGMEM = { TFT_FRAMEWDT, 154, 2, WA_CENTER };
const WidgetConfig  clockConf     PROGMEM = { 12, 176, 52, WA_RIGHT };  /* 52 is a fixed font size. do not change */
const WidgetConfig vuConf         PROGMEM = { TFT_FRAMEWDT, 100, 1, WA_LEFT };

const WidgetConfig bootWdtConf    PROGMEM = { 0, 162, 1, WA_CENTER };
const ProgressConfig bootPrgConf  PROGMEM = { 90, 14, 4 };
const BitrateConfig fullbitrateConf PROGMEM = {{DSP_WIDTH-TFT_FRAMEWDT-34, 43, 2, WA_LEFT}, 42 };

/* BANDS  */                             /* { onebandwidth, onebandheight, bandsHspace, bandsVspace, numofbands, fadespeed } */
const VUBandsConfig bandsConf     PROGMEM = { 24, 100, 4, 2, 10, 2 };

/* STRINGS  */
const char         numtxtFmt[]    PROGMEM = "%d";
[... 206 lines cut ...]
}

void DspCore::writeFillRect(int16_t x, int16_t y, int16_t w, int16_t h, uint16_t color) {
  if(_clipping){
    if ((x < _cliparea.left) || (x >= _cliparea.left+_cliparea.width) || (y < _cliparea.top) || (y > _cliparea.top + _cliparea.height))  return;
  }
  Adafruit_ILI9341::writeFillRect(x, y, w, h, color);
}

void DspCore::setClipping(clipArea ca){
  _cliparea = ca;
  _clipping = true;
}

void DspCore::clearClipping(){
  _clipping = false;
}

void DspCore::setNumFont(){
  setFont(&DS_DIGI42pt7b);
  setTextSize(1);
}
#endif

```

Krzysztof-Grzegorz, 2024-02-07:
Hello,
I also have the same problem, the display shows 3/4 of the screen
Please help, this is my first project.
[image]

dabu667, 2024-02-07:
This display is ment to work only in landscape orientation in the original code. Check myoptions.h, you probably have 90 degrees rotation.

e2002, 2024-02-07:
@TheMiro20 @Krzysztof-Grzegorz 
You specified the wrong display model in file myoptions.h

TheMiro20, 2024-02-07:
@e2002 

I am new to this. But my screen will not work when i use DSP_ST7789

Krzysztof-Grzegorz, 2024-02-07:
> @e2002
> 
> I am new to this. But my screen will not work when i use DSP_ST7789

Ooo,
I bought it with the description that it was ILI9341.
My copy comes from this store: https://pl.aliexpress.com/item/1005006110360174.html
How can I check this?

TheMiro20, 2024-02-07:
I have the same screen/product as @Krzysztof-Grzegorz.

Krzysztof-Grzegorz, 2024-02-07:
With these attempts, I think I infected this project on my computer. There are some bushes visible on the screen and no IP address
[image]

TheMiro20, 2024-02-07:
I am having the same problem with that, because i cant get to fit the whole screen, even when changing the rotation nothing happings. i have no clue what to do, the is happing with an other project. But all other test sketch work fine.

Krzysztof-Grzegorz, 2024-02-07:
That's right, other test projects worked fine until I got infected with malware. Probably from a Russian website with a description written in Word.

TheMiro20, 2024-02-07:
I am on my Macbook and nothing is wrong. No malware.

Krzysztof-Grzegorz, 2024-02-07:
> I am on my Macbook and nothing is wrong. No malware.
A bitten apple spoils quickly

Krzysztof-Grzegorz, 2024-02-07:
How can I get rid of this malware???

TheMiro20, 2024-02-07:
I have sloved it. Liked what @e2002 said. We are using the wrong DISPLAY. I think something is wrong on the Aliexpress seller side. But the my solution was to whatch the Youtube tutorial of "Daradici Levente - https://www.youtube.com/watch?v=m6ijBxIKLmo&t=279s" and to use his myconfinh.h file but to adjust the " #define DSP_MODEL           DSP_ILI9341" to ---> "#define DSP_MODEL           DSP_ST7789". This way i made my yoRadio work (ESP32-2432S028 "CYD"). Arduino IDE 1.18.1 works and Arduino IDE 2.2.1 also works mint without a problem. (P.S. Macbook Pro M1 2020)
[image]

Here is the code that is used:

'
// My youtube channel: https://www.youtube.com/@LeventeDaradici/videos
// My pastebin page: https://pastebin.com/u/LeventeDaradici
// ёRadio on Github: https://github.com/e2002/yoradio
// ёRadio - simple myoptions.h generator: https://e2002.github.io/docs/myoptions-generator.html
// Board settings: https://i.imgur.com/HdWUg0w.jpeg
// About how we load the file system on the board: https://randomnerdtutorials.com/install-esp32-filesystem-uploader-arduino-ide/
// The board I used: https://macsbug.wordpress.com/2022/08/17/esp32-2432s028/
// I bought the board from here:https://www.aliexpress.com/item/1005004502250619.html
// The answer that helped me solve the problem with the touchscreen: https://github.com/e2002/yoradio/issues/34
//
#ifndef myoptions_h
#define myoptions_h

#define DSP_MODEL           DSP_ST7789
#define DSP_HSPI            true
#define LED_INVERT          true
#define TFT_CS              15
#define TFT_DC              2
#define TFT_RST             -1
#define LED_BUILTIN         17
#define BRIGHTNESS_PIN      21
#define I2S_INTERNAL      true
#define PLAYER_FORCE_MONO     true
//#define I2S_DOUT            22
//#define I2S_BCLK            16
//#define I2S_LRC             4
#define TS_MODEL            TS_MODEL_XPT2046
#define TS_SPIPINS          25, 39, 32    /* SCK, MISO, MOSI */
#define TS_CS               33
//#define IR_PIN              27
#define LIGHT_SENSOR        34
#define AUTOBACKLIGHT_MAX   1024

#endif
`

TheMiro20, 2024-02-07:
> How can I get rid of this malware???

That is something that i am not informed of. But when something like that happend to my WINDOWS PC i always reseted my PC to the latest backup (Mostly made my pc made a backup the day before, that is how I settled it up). Anything more i cant help, sorry.

Krzysztof-Grzegorz, 2024-02-07:
Thank you,
I'm trying to compile and upload to my board

> I have sloved it. Liked what @e2002 said. We are using the wrong DISPLAY. I think something is wrong on the Aliexpress seller side. But the my solution was to whatch the Youtube tutorial of "Daradici Levente - https://www.youtube.com/watch?v=m6ijBxIKLmo&t=279s" and to use his myconfinh.h file but to adjust the " #define DSP_MODEL DSP_ILI9341" to ---> "#define DSP_MODEL DSP_ST7789". This way i made my yoRadio work (ESP32-2432S028 "CYD"). Arduino IDE 1.18.1 works and Arduino IDE 2.2.1 also works mint without a problem. (P.S. Macbook Pro M1 2020) ![tempImage2cgAKU](https://private-user-images.githubusercontent.com/139749104/303145243-8b2856ec-f323-4456-bb87-3c28e1633cab.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDczNDIyOTgsIm5iZiI6MTcwNzM0MTk5OCwicGF0aCI6Ii8xMzk3NDkxMDQvMzAzMTQ1MjQzLThiMjg1NmVjLWYzMjMtNDQ1Ni1iYjg3LTNjMjhlMTYzM2NhYi5naWY_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIwN1QyMTM5NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lMDgwYmVmOGViYzRhNTA1NzZjMGUxZWNmNDUzY2U0YTQ4NzMxNzU3ZTE0NzMwZmFhNjE5NDIyZDIyMjllNjk0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.fZrb00mnOBZy5fUYHC4z0Gyvh6Ik0ad-mMnD2fFjLAI) [ ![tempImage2cgAKU](https://private-user-images.githubusercontent.com/139749104/303145243-8b2856ec-f323-4456-bb87-3c28e1633cab.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDczNDIyOTgsIm5iZiI6MTcwNzM0MTk5OCwicGF0aCI6Ii8xMzk3NDkxMDQvMzAzMTQ1MjQzLThiMjg1NmVjLWYzMjMtNDQ1Ni1iYjg3LTNjMjhlMTYzM2NhYi5naWY_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIwN1QyMTM5NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lMDgwYmVmOGViYzRhNTA1NzZjMGUxZWNmNDUzY2U0YTQ4NzMxNzU3ZTE0NzMwZmFhNjE5NDIyZDIyMjllNjk0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.fZrb00mnOBZy5fUYHC4z0Gyvh6Ik0ad-mMnD2fFjLAI) ](https://private-user-images.githubusercontent.com/139749104/303145243-8b2856ec-f323-4456-bb87-3c28e1633cab.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDczNDIyOTgsIm5iZiI6MTcwNzM0MTk5OCwicGF0aCI6Ii8xMzk3NDkxMDQvMzAzMTQ1MjQzLThiMjg1NmVjLWYzMjMtNDQ1Ni1iYjg3LTNjMjhlMTYzM2NhYi5naWY_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIwN1QyMTM5NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lMDgwYmVmOGViYzRhNTA1NzZjMGUxZWNmNDUzY2U0YTQ4NzMxNzU3ZTE0NzMwZmFhNjE5NDIyZDIyMjllNjk0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.fZrb00mnOBZy5fUYHC4z0Gyvh6Ik0ad-mMnD2fFjLAI) [ ](https://private-user-images.githubusercontent.com/139749104/303145243-8b2856ec-f323-4456-bb87-3c28e1633cab.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDczNDIyOTgsIm5iZiI6MTcwNzM0MTk5OCwicGF0aCI6Ii8xMzk3NDkxMDQvMzAzMTQ1MjQzLThiMjg1NmVjLWYzMjMtNDQ1Ni1iYjg3LTNjMjhlMTYzM2NhYi5naWY_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIwN1QyMTM5NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lMDgwYmVmOGViYzRhNTA1NzZjMGUxZWNmNDUzY2U0YTQ4NzMxNzU3ZTE0NzMwZmFhNjE5NDIyZDIyMjllNjk0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.fZrb00mnOBZy5fUYHC4z0Gyvh6Ik0ad-mMnD2fFjLAI)
> 
> Here is the code that is used:
> 
> ' // My youtube channel: https://www.youtube.com/@LeventeDaradici/videos // My pastebin page: https://pastebin.com/u/LeventeDaradici // ёRadio on Github: https://github.com/e2002/yoradio // ёRadio - simple myoptions.h generator: https://e2002.github.io/docs/myoptions-generator.html // Board settings: https://i.imgur.com/HdWUg0w.jpeg // About how we load the file system on the board: https://randomnerdtutorials.com/install-esp32-filesystem-uploader-arduino-ide/ // The board I used: https://macsbug.wordpress.com/2022/08/17/esp32-2432s028/ // I bought the board from here:https://www.aliexpress.com/item/1005004502250619.html // The answer that helped me solve the problem with the touchscreen: #34 // #ifndef myoptions_h #define myoptions_h
> 
> #define DSP_MODEL DSP_ST7789 #define DSP_HSPI true #define LED_INVERT true #define TFT_CS 15 #define TFT_DC 2 #define TFT_RST -1 #define LED_BUILTIN 17 #define BRIGHTNESS_PIN 21 #define I2S_INTERNAL true #define PLAYER_FORCE_MONO true //#define I2S_DOUT 22 //#define I2S_BCLK 16 //#define I2S_LRC 4 #define TS_MODEL TS_MODEL_XPT2046 #define TS_SPIPINS 25, 39, 32 /* SCK, MISO, MOSI */ #define TS_CS 33 //#define IR_PIN 27 #define LIGHT_SENSOR 34 #define AUTOBACKLIGHT_MAX 1024
> 
> #endif `

---------------------
Thank you, :)
I'm trying to compile and upload to my board

TheMiro20, 2024-02-07:
> Thank you, I'm trying to compile and upload to my board
> 
> > I have sloved it. Liked what @e2002 said. We are using the wrong DISPLAY. I think something is wrong on the Aliexpress seller side. But the my solution was to whatch the Youtube tutorial of "Daradici Levente - https://www.youtube.com/watch?v=m6ijBxIKLmo&t=279s" and to use his myconfinh.h file but to adjust the " #define DSP_MODEL DSP_ILI9341" to ---> "#define DSP_MODEL DSP_ST7789". This way i made my yoRadio work (ESP32-2432S028 "CYD"). Arduino IDE 1.18.1 works and Arduino IDE 2.2.1 also works mint without a problem. (P.S. Macbook Pro M1 2020) ![tempImage2cgAKU](https://private-user-images.githubusercontent.com/139749104/303145243-8b2856ec-f323-4456-bb87-3c28e1633cab.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDczNDIyOTgsIm5iZiI6MTcwNzM0MTk5OCwicGF0aCI6Ii8xMzk3NDkxMDQvMzAzMTQ1MjQzLThiMjg1NmVjLWYzMjMtNDQ1Ni1iYjg3LTNjMjhlMTYzM2NhYi5naWY_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIwN1QyMTM5NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lMDgwYmVmOGViYzRhNTA1NzZjMGUxZWNmNDUzY2U0YTQ4NzMxNzU3ZTE0NzMwZmFhNjE5NDIyZDIyMjllNjk0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.fZrb00mnOBZy5fUYHC4z0Gyvh6Ik0ad-mMnD2fFjLAI)
> >       
> >         [
> >           
> >         
> >             ![tempImage2cgAKU](https://private-user-images.githubusercontent.com/139749104/303145243-8b2856ec-f323-4456-bb87-3c28e1633cab.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDczNDIyOTgsIm5iZiI6MTcwNzM0MTk5OCwicGF0aCI6Ii8xMzk3NDkxMDQvMzAzMTQ1MjQzLThiMjg1NmVjLWYzMjMtNDQ1Ni1iYjg3LTNjMjhlMTYzM2NhYi5naWY_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIwN1QyMTM5NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lMDgwYmVmOGViYzRhNTA1NzZjMGUxZWNmNDUzY2U0YTQ4NzMxNzU3ZTE0NzMwZmFhNjE5NDIyZDIyMjllNjk0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.fZrb00mnOBZy5fUYHC4z0Gyvh6Ik0ad-mMnD2fFjLAI)
> >           ](https://private-user-images.githubusercontent.com/139749104/303145243-8b2856ec-f323-4456-bb87-3c28e1633cab.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDczNDIyOTgsIm5iZiI6MTcwNzM0MTk5OCwicGF0aCI6Ii8xMzk3NDkxMDQvMzAzMTQ1MjQzLThiMjg1NmVjLWYzMjMtNDQ1Ni1iYjg3LTNjMjhlMTYzM2NhYi5naWY_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIwN1QyMTM5NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lMDgwYmVmOGViYzRhNTA1NzZjMGUxZWNmNDUzY2U0YTQ4NzMxNzU3ZTE0NzMwZmFhNjE5NDIyZDIyMjllNjk0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.fZrb00mnOBZy5fUYHC4z0Gyvh6Ik0ad-mMnD2fFjLAI)
> >         
> >         
> >           
> >             
> >               
> >             
> >             
> >               
> >               
> >             
> >           
> >           [
> >             
> >               
> >             
> >           ](https://private-user-images.githubusercontent.com/139749104/303145243-8b2856ec-f323-4456-bb87-3c28e1633cab.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDczNDIyOTgsIm5iZiI6MTcwNzM0MTk5OCwicGF0aCI6Ii8xMzk3NDkxMDQvMzAzMTQ1MjQzLThiMjg1NmVjLWYzMjMtNDQ1Ni1iYjg3LTNjMjhlMTYzM2NhYi5naWY_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIwN1QyMTM5NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lMDgwYmVmOGViYzRhNTA1NzZjMGUxZWNmNDUzY2U0YTQ4NzMxNzU3ZTE0NzMwZmFhNjE5NDIyZDIyMjllNjk0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.fZrb00mnOBZy5fUYHC4z0Gyvh6Ik0ad-mMnD2fFjLAI)
> >         
> >        [ ![tempImage2cgAKU](https://private-user-images.githubusercontent.com/139749104/303145243-8b2856ec-f323-4456-bb87-3c28e1633cab.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDczNDIyOTgsIm5iZiI6MTcwNzM0MTk5OCwicGF0aCI6Ii8xMzk3NDkxMDQvMzAzMTQ1MjQzLThiMjg1NmVjLWYzMjMtNDQ1Ni1iYjg3LTNjMjhlMTYzM2NhYi5naWY_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIwN1QyMTM5NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lMDgwYmVmOGViYzRhNTA1NzZjMGUxZWNmNDUzY2U0YTQ4NzMxNzU3ZTE0NzMwZmFhNjE5NDIyZDIyMjllNjk0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.fZrb00mnOBZy5fUYHC4z0Gyvh6Ik0ad-mMnD2fFjLAI) ](https://private-user-images.githubusercontent.com/139749104/303145243-8b2856ec-f323-4456-bb87-3c28e1633cab.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDczNDIyOTgsIm5iZiI6MTcwNzM0MTk5OCwicGF0aCI6Ii8xMzk3NDkxMDQvMzAzMTQ1MjQzLThiMjg1NmVjLWY [... cut ...]
> > #define DSP_MODEL DSP_ST7789 #define DSP_HSPI true #define LED_INVERT true #define TFT_CS 15 #define TFT_DC 2 #define TFT_RST -1 #define LED_BUILTIN 17 #define BRIGHTNESS_PIN 21 #define I2S_INTERNAL true #define PLAYER_FORCE_MONO true //#define I2S_DOUT 22 //#define I2S_BCLK 16 //#define I2S_LRC 4 #define TS_MODEL TS_MODEL_XPT2046 #define TS_SPIPINS 25, 39, 32 /* SCK, MISO, MOSI */ #define TS_CS 33 //#define IR_PIN 27 #define LIGHT_SENSOR 34 #define AUTOBACKLIGHT_MAX 1024
> > #endif `
> 
> Thank you, :) I'm trying to compile and upload to my board

Try it :) and if it works leave a message to know it this is the solution.

TheMiro20, 2024-02-07:
> Compilation interrupted. I think I'm missing some library.

Look in the Output monitor bellow the code.  It should say which libary your are missing.

Krzysztof-Grzegorz, 2024-02-07:
> > Compilation interrupted. I think I'm missing some library.
> 
> Look in the Output monitor bellow the code. It should say which libary your are missing.
---------------------------------
fatal error: Adafruit_ST7789.h: No such file or directory

Krzysztof-Grzegorz, 2024-02-07:
[image]

TheMiro20, 2024-02-07:
> ![Uśmiech](https://private-user-images.githubusercontent.com/105081175/303155318-2fb449f3-b6cf-4f24-b271-1a3103a349dd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDczNDUyNDMsIm5iZiI6MTcwNzM0NDk0MywicGF0aCI6Ii8xMDUwODExNzUvMzAzMTU1MzE4LTJmYjQ0OWYzLWI2Y2YtNGYyNC1iMjcxLTFhMzEwM2EzNDlkZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIwN1QyMjI5MDNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00MjI5MDMwN2I3OWJkNTI2MmRmMDE2MTM4ZWExY2M4N2RjZjJhNzEyNDAwODBkYjU5NjkwNTFiMzZhZjYyYTg2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.REEntjzQyaVt7U0u6utfml_HJSArMVBqlso4CItBs7U)

Does it work?

Krzysztof-Grzegorz, 2024-02-07:
We'll see if he logs in through this ( malware )

Krzysztof-Grzegorz, 2024-02-07:
However, this malware virus got loaded.
Blocks normal login and normal configuration.

TheMiro20, 2024-02-07:
HMMM i dont have any clue what i could be my friend.

Krzysztof-Grzegorz, 2024-02-07:
Thank you very much.
A friend in need is a friend indeed.
God bless you.

I still need to look for information about the setting of the watch and the Polish font. Łąąściżź

e2002, 2024-02-08:
@Krzysztof-Grzegorz 

> **Please help me**
> How can I get rid of this malware???

Please see Build & Flash, point 3 of this manual.
https://github.com/e2002/yoradio/wiki/How-to-flash#build--flash

TheMiro20, 2024-02-08:
> @Krzysztof-Grzegorz
> 
> > **Please help me**
> > How can I get rid of this malware???
> 
> Please see Build & Flash, point 3 of this manual. https://github.com/e2002/yoradio/wiki/How-to-flash#build--flash

If you want to flash your esp32, use the Chrome web browser and go to “ESP TOOL - https://espressif.github.io/esptool-js/”, flash it and do step by step the tutorial to install and build the esp32 radio.

Krzysztof-Grzegorz, 2024-02-08:
welcome back
1.I uninstalled arduino
2. Did a registry restore (Windows-7)
3. I installed everything from scratch.
- Arduino and recommended libraries
- I installed ESP32FS-1.1.zip

Now I get compilation errors when compiling.
-----------------------
Arduino:1.8.19 (Windows 7), Płytka:"ESP32 Dev Module, Disabled, Disabled, Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS), 240MHz (WiFi/BT), DIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None, Disabled, Disabled"

In file included from C:\Users\Studio_RCHK\Documents\Arduino\yoradio-main\yoRadio\src\displays\tools\l10n.h:14,

                 from C:\Users\Studio_RCHK\Documents\Arduino\yoradio-main\yoRadio\src\displays\displayST7789.h:14,

                 from C:\Users\Studio_RCHK\Documents\Arduino\yoradio-main\yoRadio\src\displays\dspcore.h:16,

                 from C:\Users\Studio_RCHK\Documents\Arduino\yoradio-main\yoRadio\src\core\display.h:9,

                 from C:\Users\Studio_RCHK\Documents\Arduino\yoradio-main\yoRadio\yoRadio.ino:23:

C:\Users\Studio_RCHK\Documents\Arduino\yoradio-main\yoRadio\locale\displayL10n_en.h:57:86: error: reference to 'oct' is ambiguous

   57 | const char* const mnths[]   PROGMEM = { jan, feb, mar, apr, may, jun, jul, aug, sep, oct, nov, dec };

      |                                                                                      ^~~

In file included from c:\users\studio_rchk\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-12.2.0_20230208\xtensa-esp32-elf\include\c++\12.2.0\ios:42,

                 from c:\users\studio_rchk\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-12.2.0_20230208\xtensa-esp32-elf\include\c++\12.2.0\ostream:38,

                 from c:\users\studio_rchk\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-12.2.0_20230208\xtensa-esp32-elf\include\c++\12.2.0\bits\unique_ptr.h:41,

                 from c:\users\studio_rchk\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-12.2.0_20230208\xtensa-esp32-elf\include\c++\12.2.0\memory:76,

                 from C:\Users\Studio_RCHK\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\FS\src/FS.h:24,

                 from C:\Users\Studio_RCHK\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\SPIFFS\src/SPIFFS.h:17,

                 from C:\Users\Studio_RCHK\Documents\Arduino\yoradio-main\yoRadio\src\core\config.h:6,

                 from C:\Users\Studio_RCHK\Documents\Arduino\yoradio-main\yoRadio\yoRadio.ino:20:

c:\users\studio_rchk\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-12.2.0_20230208\xtensa-esp32-elf\include\c++\12.2.0\bits\ios_base.h:1071:3: note: candidates are: 'std::ios_base& std::oct(ios_base&)'

 1071 |   oct(ios_base& __base)

      |   ^~~

C:\Users\Studio_RCHK\Documents\Arduino\yoradio-main\yoRadio\locale\displayL10n_en.h:34:12: note:                 'const char oct [8]'

   34 | const char oct[] PROGMEM = "october";

      |            ^~~

C:\Users\Studio_RCHK\Documents\Arduino\yoradio-main\yoRadio\locale\displayL10n_en.h:57:96: error: reference to 'dec' is ambiguous

   57 | const char* const mnths[]   PROGMEM = { jan, feb, mar, apr, may, jun, jul, aug, sep, oct, nov, dec };

      |                                                                                                ^~~

c:\users\studio_rchk\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-12.2.0_20230208\xtensa-esp32-elf\include\c++\12.2.0\bits\ios_base.h:1055:3: note: candidates are: 'std::ios_base& std::dec(ios_base&)'

 1055 |   dec(ios_base& __base)

      |   ^~~

C:\Users\Studio_RCHK\Documents\Arduino\yoradio-main\yoRadio\locale\displayL10n_en.h:36:12: note:                 'const char dec [9]'

   36 | const char dec[] PROGMEM = "december";
[... 33 lines cut ...]
C:\Users\Studio_RCHK\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\cores\esp32/Arduino.h:146:17: note:                 'typedef uint8_t byte'

  146 | typedef uint8_t byte;

      |                 ^~~~

Znaleziono wiele bibliotek w "WiFi.h"

Wykorzystane: C:\Users\Studio_RCHK\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\WiFi

Niewykorzystane: C:\Program Files (x86)\Arduino\libraries\WiFi

Znaleziono wiele bibliotek w "SD.h"

Wykorzystane: C:\Users\Studio_RCHK\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\SD

Niewykorzystane: C:\Program Files (x86)\Arduino\libraries\SD

exit status 1

Błąd kompilacji dla płytki ESP32 Dev Module.

Ten raport powinien zawierać więcej informacji jeśli w 
File -> Preferencje zostanie włączona opcja "Pokaż
szczegółowe informacje podczas kompilacji"

TheMiro20, 2024-02-08:
C:\Users\Studio_RCHK\Documents\Arduino\yoradio**-main**\ deleted the "-main" part. it has to to be "C:\Users\Studio_RCHK\Documents\Arduino\yoradio\

Krzysztof-Grzegorz, 2024-02-08:
Hi,
It was like that before and it was good

Krzysztof-Grzegorz, 2024-02-08:
It was like that before and it was good. Now I did it according to your suggestion and it's the same.
Here's a snippet of the error returned

In file included from C:\Users\Studio_RCHK\Documents\Arduino\yoradio\yoRadio\src\displays\tools\l10n.h:14,
                 from C:\Users\Studio_RCHK\Documents\Arduino\yoradio\yoRadio\src\displays\displayST7789.h:14,
                 from C:\Users\Studio_RCHK\Documents\Arduino\yoradio\yoRadio\src\displays\dspcore.h:16,

Krzysztof-Grzegorz, 2024-02-08:
I installed the same set on a second computer and the same errors appear. I conclude that some library is missing, but I don't know which one.

I would like to ask the creator of this project for a tip.

Krzysztof-Grzegorz, 2024-02-09:
Hello TheMiro20,
Can you tell me how many libraries you have originally loaded with Arduino? I have 19 libraries.
+ additional libraries loaded as in the photos

Am I missing any library?
[image]
[image]
[image]

TheMiro20, 2024-02-09:
When i come back home, i will send you my libary. Send me your code your file package to check yours.

Krzysztof-Grzegorz, 2024-02-09:
Okay, I'll wait until you get back. 
I just don't know what the file code is.

TheMiro20, 2024-02-09:
When come home i will look it

Krzysztof-Grzegorz, 2024-02-09:
Ok I'm waiting

Krzysztof-Grzegorz, 2024-02-09:
Puk puk - Knock knock

Krzysztof-Grzegorz, 2024-02-09:
Mr. Mirek, I need information about the original Arduino library. There are only 19 libraries in my library. I feel like I'm missing something. C:\Program Files (x86)\Arduino\libraries\

Krzysztof-Grzegorz, 2024-02-10:
So what ?
When will you come home?

TheMiro20, 2024-02-10:
@Krzysztof-Grzegorz calm down a bit, I am not your personal programmer. I have helped you with everything. Calm down, when i have the time to look and help you i will try it. Calm down!

Krzysztof-Grzegorz, 2024-02-10:
I asked because it took a while. By this time I had already dealt with it myself.

closed 2024-02-10 by TheMiro20

Krzysztof-Grzegorz, 2024-02-10:
Still, thank you for everything