e2002/yoradio #76: Bluetooth Player

stjames12, 2024-02-08:
Hello Guys, it is possible to add Bluetooth player mode except (or next to) SD card mode? As I read in the documentation when the mode is changed to SD card the wifi is turned off, maybe it is possible to play sound from bluetooth also?

bad040, 2024-02-13:
Yes. Shop for "Bluetooth emitter". Add to "line out"

stjames12, 2024-02-23:
> Yes. Shop for "Bluetooth emitter". Add to "line out"

Thank's, but I don't want to use external hardware...

phenomenonRT, 2024-06-07:
https://github.com/pschatzmann/ESP32-A2DP ?

skowalczuk, 2025-06-23:
I’d like to return to this topic and ask if anyone has managed to implement support for the built-in Bluetooth module in the ESP32-WORM within the yoRadio project?