e2002/yoradio #77: "slow stream, dropouts are possible" on this project but not on schreibfaul1's ESP32-audioI2S library

dabu667, 2024-02-09:
I know this has been discussed before but it's driving me crazy! I'm talking about the fact that on some streams (like this one: [https://1a-classicrock.radionetz.de/1a-classicrock.mp3](url)) I receive this message on console and sometimes the audio stutters. I tried everything to fix it:
 - adding PSRAM although I knew it won't help
 - improving the Wi-Fi reception
 - building the ESP32 IDF with the following parameters for better throughput:
 [image]
.. and I'm still getting the same "slow stream .."

**_BUT_**: I have just tried a simple example from schreibfaul1's ESP32-audioI2S library (version=2.0.0):
```
#include <Arduino.h>
#include "WiFiMulti.h"
#include "Audio.h"

Audio audio(true, I2S_DAC_CHANNEL_BOTH_EN); 
WiFiMulti wifiMulti;

String ssid =     "xxx";
String password = "xxx";

void setup() {
    Serial.begin(115200);
    wifiMulti.addAP(ssid.c_str(), password.c_str());
    wifiMulti.run();
    audio.setVolume(10); // 0...21
    audio.setConnectionTimeout(500, 2700);
    audio.connecttohost("https://1a-classicrock.radionetz.de/1a-classicrock.mp3");
}

void loop() {
    audio.loop();
}

void audio_info(const char *info) {
    Serial.print("info        "); Serial.println(info);
} 

```
.. and there are no warning messages or dropouts:

> info        Connect to new host: "https://1a-classicrock.radionetz.de/1a-classicrock.mp3"
> info        PSRAM found, inputBufferSize: 638965 bytes
> info        buffers freed, free Heap: 210340 bytes
> info        SSL has been established in 1338 ms, free Heap: 168696 bytes
> info        MP3Decoder has been initialized, free Heap: 145592 bytes , free stack 2796 DWORDs
> info        stream ready
> info        syncword found at pos 0
> info        Channels: 2
> info        SampleRate: 44100
> info        BitsPerSample: 16
> info        BitRate: 128000
> info        StreamTitle='Ozzy Osbourne - Shot In The Dark '
> info        StreamTitle='Them, Van Morrison - It's All Over Now, Baby Blue '
> info        StreamTitle='The Byrds - Turn! Turn! Turn! (To Everything There Is A Season) '
> info        StreamTitle='Billy Idol - White Wedding '
> info        StreamTitle='Led Zeppelin - Ramble On '

.. so I know it can be fixed. 
PLEASE help me solve this.

bad040, 2024-02-13:
I had dropouts. It was due to bad power supply. I fixed power and dropouts stopped

dabu667, 2024-02-14:
@bad040 I'm afraid it's not that.
See this: [https://github.com/schreibfaul1/ESP32-audioI2S/issues/425](https://github.com/schreibfaul1/ESP32-audioI2S/issues/425)
.. and this: [https://github.com/schreibfaul1/ESP32-audioI2S/issues/536](https://github.com/schreibfaul1/ESP32-audioI2S/issues/536), [https://github.com/schreibfaul1/ESP32-audioI2S/issues/582](https://github.com/schreibfaul1/ESP32-audioI2S/issues/582)
The author of the ESP32-audioI2S library recommends modifying some parameters in menuconfig for better throughput. I already tried that: I re-built the Arduino esp idf with modified parameters (I'm not very good at this - maybe I dis it wrong) but the problem persists.
What I did next was to use the latest version of the ESP32-audioI2S library and use a very simple sketch to play the same stream and, to my surprise there were no dropouts!
So, my guess is that there's an improvement in the latest version of the library. I tried to add it to this project (yoradio) but it seems that I'm not that good at programming.
So, what I'm asking of you guys is that you know a way to fix this.
@e2002 I know I'm asking a lot but could you integrate the latest ESP32-audioI2S library into the project? Do you think it would help with slow streams?