e2002/yoradio #78: Polish character encoding - Kodowanie polskich znaków

Krzysztof-Grzegorz, 2024-02-19:
Hello, the author of this internet radio.
I have an urgent request for help in arranging Polish letters with tails.
----------------------------------------------------------------------------
Witam autora (eRadio) tego radia internetowego.
Mam gorącą prośbę o pomoc w ustawieniu polskich liter z ogonkami.

Ą - Ć - Ę - Ł - Ń - Ó - Ś - Ź - Ż - ą - ć - ę - ł - ń - ó - ś - ź - ż

Krzysztof-Grzegorz, 2024-02-20:
(https://github.com/e2002/yoradio/assets/105081175/51d18072-8092-45e8-bf94-2d77e0ad0b12)

pszkola, 2024-03-27:
What's display type? If is't Nextion it might be a Character Encoding type PL - > ISO 8859-2 or if different HMI it might be metter of glcdfont.c not encoding PL letters.

Krzysztof-Grzegorz, 2024-03-27:
Hello, 
I have two displays, one is ILI9341 and the other is ST7789

pszkola, 2024-03-28:
Krzysztof, 
look at link below it might be same issue, or if not search "glcdfont.c polskie znaki" in Google. There are some interesting articles. 

https://forbot.pl/forum/topic/19053-funkcja-zmieniajaca-czcionke-z-biblioteki-adafruit-gfx-nie-dziala/