e2002/yoradio #79: const uint8_t volumetable[22]

DonEugraf, 2024-02-20:
can anyone help me please?
I'm trying to make volume 0-21 instead of 0-254
I have declared the volume table and modified everywhere instead of vol 254 to 21 
it works fine with web and remote but when using encoder it circles back to 0 if 21 is exceeded and back to 21 if going below 0.
I've spent weeks trying to figure out why this is happening without success, if you guys made it work could you share what you changed please?

coccobello27, 2024-09-01:
Have found solution for remaining volume?

DonEugraf, 2025-02-16:
I have documented all the changes to make it work with 21 instead of 254. if anyone can make it more elegant would be nice. Beware of the bug or circling around on encoder !!!

/////////////////////////////		Volume to 22 steps		////////////////////////

\yoRadio\src\core\player.cpp lane 294
void Player::stepVol(bool up) {
  if (up) {
    if (config.store.volume <= 21 - config.store.volsteps) {
      setVol(config.store.volume + config.store.volsteps);
    }else{
      setVol(21);

\yoRadio\src\core\player.cpp lane 310
uint8_t Player::volToI2S(uint8_t volume) {
  int vol = map(volume, 0, 21 - config.station.ovol * 3 , 0, 21);
  if (vol > 21) vol = 21;

\yoRadio\src\core\telnet.cpp lane 227
if (volume > 21) volume = 21;

\yoRadio\src\core\netserver.cpp lane 982
if (v > 21) v = 21;

\yoRadio\src\core\controls.cpp lane 88
encoder.setBoundaries(0, 21, true);

\yoRadio\src\core\controls.cpp lane 94
encoder2.setBoundaries(0, 21, true);

\yoRadio\src\core\controls.cpp lane 173,458
if(nv>21) nv=21;

\yoRadio\src\core\display.cpp lane 117
    _volbar = new SliderWidget(volbarConf, config.theme.volbarin, config.theme.background, 21, config.theme.volbarout);

\yoRadio\src\audioI2S\AudioEx.h lane 449
//const uint8_t volumetable[22]={   0,  1,  2,  3,  4 , 6 , 8, 10, 12, 14, 17,
//                                     20, 23, 27, 30 ,34, 38, 43 ,48, 52, 58, 64}; //22 elements
    const uint8_t volumetable[22] = {0, 2, 4, 8, 12, 16, 24, 32, 40, 50, 64, 80, 100, 120, 140, 160, 180, 200, 210, 235, 235, 254};

\yoRadio\src\audioI2S\AudioEx.h lane 500
uint8_t         m_vol=21;                       // volume

//////	if you use VS1053
find the abobe on yoRadio\src\audioVS1053\audioVS1053Ex.h lane 237

\yoRadio\src\audioI2S\Audio.cpp lane 1730,1983
if(tmp > 21) tmp = 21;

	lane 4665
//---------------------------------------------------------------------------------------------------------------------
void Audio::setVolume(uint8_t vol) { // vol 22 steps, 0...21
//    if(vol > 254) vol = 254;
//    m_vol = vol;
    if(vol > 21) vol = 21;
    m_vol = volumetable[vol];
}
//---------------------------------------------------------------------------------------------------------------------
uint8_t Audio::getVolume() {
//    return m_vol;
    for(uint8_t i = 0; i < 22; i++) {
        if(volumetable[i] == m_vol) return i;
    }
    m_vol = 12; // if m_vol not found in table
    return m_vol;
}
//---------------------------------------------------------------------------------------------------------------------

		lane 4688
float step = (float)m_vol *21/254;

//////	if you use VS1053 find the above on \yoRadio\src\audioVS1053\audioVS1053Ex.cpp 

\yoRadio\data\www\index.html lane 61
<input type="range" id="volrange" class="slider" name="volume" data-slaveid="volinfo" min="0" max="21" value="0">