e2002/yoradio #80: All radio stations freeze after a while....

fratellos, 2024-02-24:
The project is a great achievement, congratulations!
I made it myself, with the help of the members of an electronics forum in my country. I use an ESP-32-VROOM-32D, a 1.8" display with ST7735 controller (128x160 RGB TFT LCD), a PCM5102A module (not connected yet!).
It works... and not really! Invariably, any type of radio station I would like to listen to (mp3, aac, aacp), after an undefined, variable time, the station freezes and nothing happens. ESP works, the weather information, the time are shown ... but not the radio. I left only one mp3 in the playlist, I thought it was a memory problem. But it also got stuck after a while...
I have no problems with the WiFi network; I constantly monitor the network and have no interruptions.
Thank you in advance for any solution suggestion!

[image]

dabu667, 2024-03-02:
Check the serial output when it happens

ESPNoodle, 2024-11-17:
Try removing the "s" on any Https: addresses.... 😎👌 So just, http:....