e2002/yoradio #81: Rotated display on ESP32-2432S028 (CYD)

blit27, 2024-02-24:
I purchased this item when it was recently on sale:
[https://www.aliexpress.us/item/3256805849164942.html](https://www.aliexpress.us/item/3256805849164942.html)
During compilation I used settings provided in the Wiki for the ESP32-2432S028. The result of using DSP_ILI9341 was a rotated display as shown on the left in the picture below.
If I use DSP_ST7789 in myoptions.h all is as expected (see pic on right). There appears to be some variants in ESP32-2432S028's. Caveat emptor!
[image]

hmfuchs, 2024-05-07:
It’s the Same by me. All new boards i have only working with DSP_ ST7789 in my options.

onkel-joerg, 2024-07-27:
do you have a working myoptions.h for me?
I got an error when i change the "DSP_ILI9341" to "DSP_ ST7789"
(myoptions.h:14:34: error: missing binary operator before token "ST7789")

I'm a noob I have no idea what a missing binary operator means.

blit27, 2024-07-27:
Examine your entry, there should be no spaces in "DSP_ST7789"

zalves, 2024-08-14:
Hello @blit27 cna you share your myoptions.h, at least the ones relative to the display and touch screen? I've tried to configure, but I always get a black screen with the ST7789.

These are myoptions:

#define DSP_MODEL			DSP_ST7789
#define DSP_HSPI			true
#define TFT_DC			2
#define TFT_CS			15
#define I2S_DOUT			21
#define I2S_BCLK			22
#define I2S_LRC			27
#define TS_MODEL			TS_MODEL_XPT2046
#define TS_SPIPINS          25, 39, 32    /* SCK, MISO, MOSI */
#define TS_CS			33
#define SDC_CS			5

onkel-joerg, 2024-08-15:
what happens when you touch your screen?
or click on play in the web-ui?

blit27, 2024-08-15:
zalves, your dear uncle has good troubleshooting suggestions.
Here is my 'myoptions.h' and a photo of the CYD box that it works with.

#define DSP_MODEL			DSP_ST7789
#define DSP_HSPI			true  // HSPI (SCK-14, MISO-12, MOSI-13)
#define TFT_CS      15
#define TFT_DC			2
#define TFT_RST     -1
#define BRIGHTNESS_PIN      21
#define PLAYER_FORCE_MONO           true
#define I2S_DOUT		4
#define I2S_BCLK		16
#define I2S_LRC			17
#define TS_MODEL		TS_MODEL_XPT2046
#define TS_CS			33
#define TS_SPIPINS      25, 39, 32    /* SCK, MISO, MOSI */
#define LIGHT_SENSOR        34
#define AUTOBACKLIGHT_MAX   1024
//#define SD_HSPI  false  // VSPI (SCK-18, MISO-19, MOSI-23)
//#define SDC_CS  5

[image]

onkel-joerg, 2024-08-15:
> zalves, your dear uncle has good troubleshooting suggestions. Here is my 'myoptions.h' and a photo of the CYD box that it works with.
> 
> #define DSP_MODEL DSP_ST7789 #define DSP_HSPI true // HSPI (SCK-14, MISO-12, MOSI-13) #define TFT_CS 15 #define TFT_DC 2 #define TFT_RST -1 #define BRIGHTNESS_PIN 21 #define PLAYER_FORCE_MONO true #define I2S_DOUT 4 #define I2S_BCLK 16 #define I2S_LRC 17 #define TS_MODEL TS_MODEL_XPT2046 #define TS_CS 33 #define TS_SPIPINS 25, 39, 32 /* SCK, MISO, MOSI */ #define LIGHT_SENSOR 34 #define AUTOBACKLIGHT_MAX 1024 //#define SD_HSPI false // VSPI (SCK-18, MISO-19, MOSI-23) //#define SDC_CS 5
> 
> ![cyd_box](https://private-user-images.githubusercontent.com/89953615/358229474-bc0722d3-0322-493f-924b-cea628421665.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM3Mzc1OTksIm5iZiI6MTcyMzczNzI5OSwicGF0aCI6Ii84OTk1MzYxNS8zNTgyMjk0NzQtYmMwNzIyZDMtMDMyMi00OTNmLTkyNGItY2VhNjI4NDIxNjY1LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODE1VDE1NTQ1OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdkZDMyZGRjOTJmNzc1Y2I4NTdhYTg3NDEyMGIwMTg3MzI3ZmE0YTQ3NmIwMWI1NzVlM2FlMTc0MDNiZDM0YmImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Pz5fgHoYw1rQy9NO670t-tzWzmcnotX6XNfCTLtOHEI)

this version works for me
here my CYD:
[image]