e2002/yoradio #82: ESP32-3248S035C touchscreen does not work

dorinik, 2024-02-28:
Thanks for your work!
I try to use ESP32-3248S035C, but the touchscreen does not work.
I tested the touchscreen with the demo file and it works!
I use the settings:
#define DSP_MODEL           DSP_ST7796
#define DSP_HSPI            true
#define LED_INVERT          true
#define TFT_CS              15
#define TFT_DC              2
#define TFT_RST             -1
#define BRIGHTNESS_PIN      27
#define SDC_CS              5
#define I2S_DOUT            17
#define I2S_BCLK            4
#define I2S_LRC             16
#define TS_MODEL            TS_MODEL_GT911
#define LIGHT_SENSOR        34
#define AUTOBACKLIGHT_MAX   1024

[image]

Best regards!

DonEugraf, 2024-02-28:
Hi,
Have you tried all 3 options?
/*        TOUCH SCREEN            */
#define TS_MODEL_UNDEFINED      
#define TS_MODEL_XPT2046        
#define TS_MODEL_GT911

dorinik, 2024-02-29:
Hi DonEugraf,
I just tried:
#define TS_MODEL_GT911
Tomorrow I will try the other 2 options.
Thank you for the advice !
I hope one of the options will work.

dorinik, 2024-03-02:
I tested the other 2 options for TS_MODEL, but they don't work.
It should work with GT911 because the chip on the touch wiring is GT911...
I think the library for GT911 is not ok.

[image]

DonEugraf, 2024-03-02:
#define TS_HSPI?
on mine i have define as follow, i can't see above setting on yours
```
/* !!! Do not be alarmed, all right, the default values ​​are not included in this file !!! */
#define DSP_MODEL			DSP_ST7735  //DSP_SSD1306 //DSP_DUMMY//DSP_2004I2C
#define DTYPE INITR_BLACKTAB // 1.8' https://aliexpress.ru/item/1005002822797745.html
#define BRIGHTNESS_PIN      22
#define TFT_CS            5                 /*  SPI CS pin  */
#define TFT_RST           -1                /*  SPI RST pin.  set to -1 and connect to Esp EN pin */
#define TFT_DC            4                 /*  SPI DC/RS pin  */
#define DSP_HSPI          false             /*  Use HSPI for display  */
/*  VSPI PINS. SCL(SCK, CLK) must be connected to pin 18
               SDA(MOSI, DIN, SDI) must be connected to pin 23  */
/*  HSPI PINS. SCL(SCK, CLK) must be connected to pin 14
               SDA(MOSI, DIN, SDI) must be connected to pin 13  */
#define TS_MODEL              TS_MODEL_XPT2046  //TS_MODEL_UNDEFINED//TS_MODEL_GT911
#define TS_HSPI   false      // use HSPI for touchscreen (miso=12, mosi=13, clk=14) instead of VSPI (by default)
#define TS_CS      32
#define SD_HSPI   false				 /* use HSPI for SD (miso=12, mosi=13, clk=14) instead of VSPI (by default)	*/
#define SDC_CS      33
/******************************************/
```

dorinik, 2024-03-03:
My board has capacitive touch, not resistive like yours.
The pins of my touch (GT911) are INT=21, RST=25, SCL=32, SDA=33, the pins of your touch (XPT2046) have CLK, MOSI, MISO, CS, IRQ.
The pins for GT911 are defined in options.h (208-218).

[image]

[image]

[image]

DonEugraf, 2024-03-03:
I understand but you still have to define it as true or false I suppose

echosmart792, 2024-09-13:
https://youtu.be/MSIvnyaKNxk

https://satelit-info.com/phpBB3/viewtopic.php?f=172&t=3380&p=26183#p26178

`#ifndef myoptions_h
#define myoptions_h
//
// Video on youtube: https://youtu.be/MSIvnyaKNxk
//
#define DSP_MODEL           DSP_ST7796
#define DSP_HSPI            true
#define TFT_DC          2
#define TFT_CS          15
#define BRIGHTNESS_PIN          27
#define I2S_INTERNAL            true
#define PLAYER_FORCE_MONO           true
#define TS_MODEL            TS_MODEL_GT911
 
#endif`