e2002/yoradio #83: Nach Start kurz Hauptbild danach nur "Volume" zu sehen.

fetzer91, 2024-02-29:
[image]

DonEugraf, 2024-03-02:
Use different pins for your encoder

fetzer91, 2024-03-03:
Ich habe keinen Encoder angeschlossen. Am ESP32 hängen nur das Display und der VS1053. Im Anhang sende ich meine myoption.h.

#ifndef myoptions_h
#define myoptions_h

/* Generated by myoptions.h generator - v0.8.1 */
/* https://e2002.github.io/docs/myoptions-generator.html?r=71,1,2,5,8,19,41,44,48,60,62&i=1,2,3,4,18,19,20,21,34&v=21,22,-1,255,33,4,36,-1,5 */
/* Put this file next to yoRadio.ino and restart Arduino IDE. */

/* !!! Attention !!! */
/* This configuration depends on some additional libraries. Please install them using the library manager: */
/* Adafruit GFX Library, Adafruit ILI9341, XPT2046_Touchscreen */

/* !!! Do not be alarmed, all right, the default values ​​are not included in this file !!! */

#define DSP_MODEL			DSP_ILI9341
#define TFT_DC			21
#define TFT_CS			22
#define I2S_DOUT			255
#define VS_HSPI			true
#define VS1053_CS			33
#define VS1053_DCS			4
#define VS1053_DREQ			36
#define TS_MODEL			TS_MODEL_XPT2046
#define TS_CS			5

#endif
[image]

DonEugraf, 2024-03-03:
Gpio 5 is also used for vspi c0 so use TS_CS 13 or something