e2002/yoradio #93: flip the screen ILI9488 3.5"

RuslanRadioman, 2024-05-05:
How to flip an image 180 degrees? Team
#define TFT_ROTATE 3
written in the myoptions.h file does not work. I tried different values from 0 to 3.
[image]

JulesDeChezSmithEnFace, 2024-05-06:
modify  src\displays\displayILI9488.cpp
in section  DspCore::initDisplay and add    **setRotation(2);**

you should modify also   \src\displays\conf\displayILI9488conf.h to relocate items to the right place

RuslanRadioman, 2024-05-07:
in the DspCore::initDisplay section added setRotation(3);
Problem solved.
[image]

closed 2024-05-07 by RuslanRadioman