e2002/yoradio #98: Feature request: Chande

onkel-joerg, 2024-07-28:


hevet, 2024-07-28:
And who is supposed to fulfill it when the project is unfortunately in the abandonment phase?

onkel-joerg, 2024-07-29:
Oh, I apologize.
I didn't realize that this project was marked as "in the abandonment phase".
Where is this status visible so I can avoid such harassment in the future?

hevet, 2024-07-29:
I meant more that nothing interesting has happened on github for 10 months, no new version or bug fixes. There is a topic on the forum or a group on facebook, maybe look for help there, some people modify the version "for themselves", maybe someone did something like that. I made up the status, in no way was it supposed to discourage you from writing or asking for help, but unfortunately here on github I doubt you will get help.