e2002/yoradio #101: 'network' was not declared in this scope; did you mean 'Network'

Duncan-Hill, 2024-09-01:
I see some else experiencing similar issue, is this a bug or an issue with the way I'm setting up the system?

In function 'void setup()':

yoRadio:39:3: error: 'network' was not declared in this scope; did you mean 'Network'?
39 |   network.begin();

dronix27, 2024-09-05:
> I see some else experiencing similar issue, is this a bug or an issue with the way I'm setting up the system?
> 
> In function 'void setup()':
> 
> yoRadio:39:3: error: 'network' was not declared in this scope; did you mean 'Network'? 39 | network.begin();

The board version should not be higher 2.0.17

echosmart792, 2024-09-13:
Downgrade ESP32 board manager from Arduino IDE to version 2.0.17.

Ivanyushenkov, 2024-09-15:
Да, подтверждаю. При понижении менеджера платы ESP32 из Arduino IDE до 2.0.17 скетч стал нормально компилироваться.

DaradiciLevente, 2024-09-24:
Я так рад, что это работает для вас.

Clumpton, 2024-09-26:
> Downgrade ESP32 board manager from Arduino IDE to version 2.0.17.

This definitely works! Just been through the same experience.

e2002, 2024-11-27:
- fixed in version [0.9.337b](https://github.com/e2002/yoradio?tab=readme-ov-file#v09337b)