e2002/yoradio #104: It seems to be very COMPLICATED

ovidiu169, 2024-10-27:
Hello everybody !

After I read many times the instructions, it is not very clear how it supposed to build this radio.
The Code is not compiling, the data is not uploading to the board. (ESP WRover module)
After I uploaded all the resources the Arduino IDE 1.8.19 is saying that the files are not found !!!
[image]
[image]

ovidiu169, 2024-10-27:
I found one of the secrets !!!
Degrading the ESP 32 Library version until 2.0.3 !!!
I succeeded to upload the code !!!

closed 2024-10-27 by ovidiu169

ESPNoodle, 2024-11-17:
Find us on Facebook! ...  Yoradio