e2002/yoradio #105: Cannot compile  for JC4827W543C on platformIO

bobcroft, 2024-10-30:
Not matter what I try I keep getting this compiler error, it happens for most files that are being compiled. can someone helps me please.
xtensa-esp32-elf-g++: error: unrecognized command-line option '--abbrev=8'
xtensa-esp32-elf-g++: error: unrecognized command-line option '--dirty'
xtensa-esp32-elf-g++: error: unrecognized command-line option '--always'
xtensa-esp32-elf-g++: error: unrecognized command-line option '--tags)"'
xtensa-esp32-elf-g++: error: unrecognized command-line option '--get'

Also would anyone be kind enough to post notes on what is required to be added to platformIO from the Github files.  I had assumed everything.  Ideally if anyone could share a working platformIo set up that would be greatly appreciated.
TIA
Bob

DaradiciLevente, 2024-12-27:
Did you manage to set up the myoptions.h file for this microcontroller in the Arduino IDE? I have this controller too, but I haven't had time to try using it for yoRadio yet. Can the audio part that comes with this microcontroller be used in yoRadio?
I would be happy if you could share the display and audio settings for this microcontroller with us.

bobcroft, 2024-12-28:
Hi, sorry for the delay is responding.  I didn’t do anything with YoRadio in the Arduino IDE.  I’ll forward your email to coolMd, who has, I believe, had some success with YoRadio on 543.  I don’t think he got it completely working.

Bob

CoolmdXi, 2024-12-28:
Dear Bob and DaradiciLevente,
I did manage to get the sketch to upload and run with alittle bit of effort using the fork from the original sketch. https://github.com/dasdgw/yoradio.
I changed some of the settings and condensed the required code and it works on the JC4827W543C. I have used it using the boards onboard DAC as well as external DAC and it works ok.
I will attach the files I used and also the myoptions files for you to try. It will not allow me to attach files and folders to this discussion. will have to figure out another way for you to get the files I used

CoolmdXi, 2024-12-28:
[image]

CoolmdXi, 2024-12-28:
these are myotions.h , cant remember if i had to do much else . It worked and compiled in the latest Arduino IDE and didn't need to use older Arduino or board definitions.... didn't get the volume and wifi font correct but all else worked. Don't forget to upload data as you would with the original sketch.
Hope this helps
Coolmd

CoolmdXi, 2024-12-28:
I did cut and move a few things around but not sure how to share my version which worked with the 543C board....too big for pastebin, maybe I could use google drive..i will look into it

CoolmdXi, 2024-12-28:
https://drive.google.com/drive/folders/1AFv9cP9gCytROd7ar_A1jNJ6kK-TSDcD?usp=sharing

DaradiciLevente, 2024-12-28:
The file on Google Drive does not have access to download that file. Thank you for making your documentation available to us! I will try to recreate it from the information you have sent us so far and if you give us access to the link in Google Drive I will try to use the information there as well. Thanks again for your kindness in helping us and I wish you Happy Holidays!

CoolmdXi, 2024-12-28:
https://drive.google.com/drive/folders/1NPAPASQaL7rOXfno-fxQ9vwwZTQqLg_p?usp=sharing

CoolmdXi, 2024-12-28:
Could you please try this one, maybe it will work better. if it does or does not can you let me know please?

DaradiciLevente, 2024-12-28:
Thank you very much! The link works. I will try to upload it to the microcontroller.

CoolmdXi, 2024-12-28:
I am hopefull it should work. Please let me know..good luck

DaradiciLevente, 2024-12-28:
I managed to upload the code, but nothing seems to appear on the display. This is what my display looks like after uploading yoRadio.
[image]

In the Arduino IDE I made the following settings:
[image]

The microcontroller has yoRadio installed, version v0.9.300.
I added the channel list, Wifi settings, everything seems OK, except that I did not connect a speaker to the microcontroller's speaker connector, because I do not have that connector at home and the documentation does not show me which connector is needed there.
Unfortunately, the display shows what I showed you in the first photo.
The one in blue seems to be the VUmeter because it moves continuously like a VUmeter.

DaradiciLevente, 2024-12-28:
The mistake is mine!
My module is called JC3248W535C and not JC4827W543 as this thread says. I will probably have to find the pinout for this device somewhere.

My device is bought from here:
https://www.aliexpress.com/item/1005007566046827.html

bobcroft, 2024-12-29:
Hi, I note you are using the 535 not the 543.  I do not know if the 535 uses a different LCD driver to the 543 but looking at the 1st photo you posted I would guess it is a screen driver issue.  I do have a couple of 535’s but I haven’t had time to try them out yet.  If you need any help testing please ask.

vu.jpg (view on web)<https://github.com/user-attachments/assets/a6c3bb08-a9fa-48fe-b6e0-5e98b64877b5>

IDE33.png (view on web)<https://github.com/user-attachments/assets/283fc2dd-1215-4dda-a240-971c9e443ecd>

CoolmdXi, 2024-12-29:
I am not sure what display driver the 535 board uses, or its dac pins etc. I dont have the information on that board, if it uses a similar display driver to the 543C it may just be pin definitions that need altering. 
Bob did you download the manufacturers Files for the 535 Board?

CoolmdXi, 2024-12-29:
Dear Daradici,
The Gpio connectors on the 543C board are JST 1.25mm connectors. I am not sure but I think the speaker connector is the same. or maybe a 1.0mm JST. I did buy some from Ali express, I will have a look and find out what they are.

CoolmdXi, 2024-12-29:
I downloaded the manufacturers instruction and it uses a AXS15231B display driver. There are no demo/examples using any normal graphics library at all only LVGL so I am not sure what library would support it . A bit more investigation reveals that the driver is used by lilygo t display s3 long, which is supported in Larry Banks bb_spi_lcd library and also Arduino GFX, but both have a different display resolution compared to the 535 board you have. Not having a board, would make it hard to attempt to get it to work and I dont really want to buy one as there seems to be literally no support for it.

bobcroft, 2024-12-29:
Hi Mark,

I can’t answer your questions fast enough!! I did find a file for that display but it is garbage, a lift from the 543 file I think.  I seem to recall Larry making some comment on the AXS15231B driver a while ago.  I am looking into the AXs driver in a moment.  I completely agree about support and it is much the same for the 550.

It is, as we have said before, these displays are fine if one wants to use the Arduino-GFX library upon which the examples are based.  That is great for new projects any individual wants to write.  It is not so great if one is trying to port projects like YoRadio.
I dare say the 535 can be made to work with Larry’s libraries with time and effort but sadly I don’t have the time at the minute.  I am wrestling with the 550 and I am so close but the bloody screen is just black on platformIO but it works OK on duino if one can wait ages for it to compile.
I just got some code that is supposed to run on a 550 but it doesn’t compile and it uses LVGL which is another crock of shit that breaks with every version update.

Have a good day.

Bob

DaradiciLevente, 2025-01-28:
Hi! I also bought the 4.3 inch version https://vi.aliexpress.com/item/1005006729377800.html but although last time the 3.5 inch one compiled without errors and was loaded onto the board, now it doesn't want to compile, I have a lot of compilation errors.

DaradiciLevente, 2025-01-28:
I managed to compile and upload the code to the microcontroller, yoRadio works but I have a white display. This is the myoptions.h file used: `#ifndef myoptions_h
#define myoptions_h

#define L10N_LANGUAGE     EN                

#define DSP_MODEL         DSP_NV3041A

/******************SCREEN SETUP************************/

#define TFT_CS  45                  
#define TFT_RST -1                
#define TFT_SCK 47                
#define TFT_D0  21
#define TFT_D1  48
#define TFT_D2  40
#define TFT_D3  39
#define GFX_BL 1 //############################

/******************************************/

#define PLAYER_FORCE_MONO false    // true for onboard DAC

//#define I2S_INTERNAL      false             /*  If true - use esp32 internal DAC  */

/*        I2S DAC                 */
#define I2S_DOUT         41                 /*  5,14,9 external DAC...41 42 2 internal dac to speaker output*/
#define I2S_BCLK         42                 /*  BCLK Bit clock */
#define I2S_LRC          2                /*  WSEL Left Right Clock */
/******************************************/

//#define RSSI_STEPS       -50,-60,-70,-80

#define NO_ICONS false      // Set to true to disable icons and use text instead ###############################
#define RSSI_DIGIT true   // Set to true to display RSSI as digits instead of bars############################

/*  TOUCHSCREEN  */
#define  TS_MODEL   TS_MODEL_GT911

/*  Capacitive I2C touch screen  */

#define TS_SDA      8    //      //was ts_sda etc
#define TS_SCL      4    //  ####################    
#define TS_INT      3    //   ##################   
#define TS_RST      38 

   
/******************************************/

//#define MUTE_PIN          255               /*  MUTE Pin */
//#define MUTE_VAL          HIGH              /*  Write this to MUTE_PIN when player is stopped */
//#define BRIGHTNESS_PIN    255               /*  Pin for adjusting the brightness of the display (output 0 - 3v3) */

#endif
`

CoolmdXi, 2025-01-28:
Dear Daradici,
It might be worth Rolling back the board manager to 3.01 and I i know that very recently Arduino_GFX updated so I would set that back to version 1.4.9.
 I would probably try Arduino_GFX library first. DO you have a copy of your compile errors?

DaradiciLevente, 2025-01-28:
Is it possible that the display model on this microcontroller has changed?

[image]

CoolmdXi, 2025-01-28:
Arduino_GFX is probably the issue, I tried to do something with another sketch about 5 days ago, and had to rollback the version to 1.4.9 otherwise i got alot of compile errors

CoolmdXi, 2025-01-28:
Let me know if he have any luck?

DaradiciLevente, 2025-01-28:
I changed the Arduino GFX library to the recommended version, but after compiling without errors and loading the code on the microcontroller I have a white screen. yoRadio is running in the background, but it doesn't show anything on the display.

CoolmdXi, 2025-01-28:
DID you use my files from onedrive?

DaradiciLevente, 2025-01-28:
Yes, I'm using the same code you gave me via Google Drive. It loads on the microcontroller but doesn't display anything. What I see on the display is like the vumeter moving according to the sound on the radio channel.

[image]

CoolmdXi, 2025-01-28:
I don't have a board to retest it on at the moment.  Did you rollback esp32 boards as well?

CoolmdXi, 2025-02-12:
Dear Daradici, I have revisited the sketch and folders and have been able to reflash the yoradio onto the 543C board. I managed to get it to work, but with only a certain set of library versions and ESP32 board settings. If you are still interested let me Know

DaradiciLevente, 2025-02-16:
Of course I'm interested! I hope we can somehow get it to work on this board too.

CoolmdXi, 2025-02-17:
Dear Daradici,
I will add a link for you to see code.
The my options is set to use the onboard DAC and Amp on the 543C board as the audio quality is very very good.
It is important to still upload "SPIFFS" and you can now use the latest arduino 2.3.4 version.
It is very important to use the libraries and versions as listed at the top of the main .ino file.
I will list them here

/* must use AsyncTCP version 1.1.3
ESP Async Webserver 2.9.0
GFX Arduino 1.4.7
ESP32 boards 3.0.1 

https://drive.google.com/drive/folders/1H5U3VFy37TpoulWnPiBWE_YEuvBKGuki?usp=drive_link

Please let me know how you get on, hopefully will make a great video..

bobcroft, 2025-02-17:
Hi Mark,

Thanks for linking me in to this post.  I’ll try your version of yoRadio as soon as I get a   minute.

Bob

https://drive.google.com/drive/folders/1H5U3VFy37TpoulWnPiBWE_YEuvBKGuki?usp=drive_link

https://drive.google.com/drive/folders/1H5U3VFy37TpoulWnPiBWE_YEuvBKGuki?usp=drive_link

DaradiciLevente, 2025-02-17:
Thanks for the information! For now, I won't have time to try it today, but in the next few days I will come back to the subject and try it exactly as you described. Thank you very much for the time you give me to complete this project.

DaradiciLevente, 2025-02-21:
[image]

First positive results thanks to your help!
yoRadio works, the sound played by the audio output is very good, the change between channels is very fast, it moves excellently!

But I have a small problem. The clock is not displayed, only the seconds.
I changed a few skins thinking that the problem comes from the skin, but for now the clock cannot be displayed on the display. Another problem would be the RSSI indicator and the VOLUME indicator, which are not indicated correctly, even if I changed the font file in the Adafruit GFX library.

CoolmdXi, 2025-02-21:
I am glad you got it to work. I don't know why the clock is not displaying mine displays correctly . The volume and WiFi are incorrect . I never did try to fix this as was happy to get it to work!I would be interested to find out if you can get these problems fixed. If I find out anything I will let you know.RegardsMark D

  1.jpg (view on web)

1.jpg (view on web)

DaradiciLevente, 2025-02-23:
[image]

Thanks to you, I solved the problem!
Except that the external fonts don't work for me, so I have to change the external fonts to default fonts, adjust their size so that they are displayed in their places.
I solved the problem with the volume icon and the RSSI indicator icon by changing the icons to text. For the moment, I haven't found another solution.

I'll add my change to GitHub so that anyone can have it available when they encounter the problems I encountered.

Thanks again for the help!

CoolmdXi, 2025-02-23:
I am pleased that this is now working for you. I am glad that I could help, Could you explain here how you got the corrected fonts and displays as I am not sure how to find it in github, I dont really understand branches and pulls etc.
Many Thanks
Mark

DaradiciLevente, 2025-02-23:
For the RSSI indicator, edit the file: yoRadio/src/core/display.cpp like this:

[image]

[image]

DaradiciLevente, 2025-02-23:
To replace the VOL icon with text, edit the file:
yoRadio/src/displays/conf/displayNV3041Aconf_custom.h
like this:

[image]

We replace the line:
const char         voltxtFmt[]    PROGMEM = "\023\025%d";
with the line:
const char         voltxtFmt[]    PROGMEM = "VOL %d";

DaradiciLevente, 2025-02-23:
I have posted a modified version on github with the firmware you provided and my modifications.
https://github.com/DaradiciLevente/yoRadio_on_JC4827W543C

DaradiciLevente, 2025-03-05:
I made a video showing how we can upload this code to this microcontroller in a very fast way!
- Without having to choose various libraries in the Arduino IDE.
- The upload type to the microcontroller only takes a few seconds!

https://youtu.be/828zJZ32_Bs

DaradiciLevente, 2025-03-05:
We also added the SD card data.
Now we can also play mp3 files from a micro SD card that we insert into the SD card reader on this board.
You can see a short video here:
https://youtu.be/pMsSwy1TMJo

CoolmdXi, 2025-03-05:
Many thanks I have never bothered to use the sd card. I will give it a try. Thanks again

https://youtu.be/pMsSwy1TMJo

bobcroft, 2025-03-06:
Excellent progress and the idea to add the ability to play files from the SD card is brilliant.  I also think the upload method you provided is the best option because it will work with all systems, windows, Mac, Linux and it is straight forward without using web resources.  It is usable by all people whether they are new to these display boards or not.  In any case a web flasher can be used with the available .bin file by anybody who is so minded.
I do think you should monetize a future versions or upgrades as you have put a lot of work into this project, but this is only my view.

nvalw, 2025-03-06:
[image]
 [@CoolmdXi](https://github.com/CoolmdXi)  Thank you for the files you sent!

CoolmdXi, 2025-03-08:
> [image] [image] [image]
> 
> [yoradio_on_JC4827W543C_0.9.434.zip](https://github.com/user-attachments/files/19144084/yoradio_on_JC4827W543C_0.9.434.zip) - full binary files with SPIFFS. Yoradio 0.9.434 v.0.9.434 on JC4827W543C - checking... Includes RUS and EN versions.

Did you correct the clock fonts or did they work for you from sketch?

nvalw, 2025-03-08:
I just applied them correctly for this module... Of course, they are not themselves... And the interface language certainly has nothing to do with it.

CoolmdXi, 2025-03-08:
Could you show on here how to correctly install the clock and date FONTS as some peoples do not work. The font you are using is the one the original sketch had in it, but it didnt appear to work for everybody?

nvalw, 2025-03-12:
[image]

[image]

Joslen-Bomound, 2025-03-27:
Дякую автору за розробку. Потрібна  версія з 5102 з російською мовою.

Joslen-Bomound, 2025-04-01:
> З новими шрифтами. [image]
> 
> [image]
> 
> [yoradio_on_JC4827W543C-RUS-0x00000.zip](https://github.com/user-attachments/files/19217294/yoradio_on_JC4827W543C-RUS-0x00000.zip)

Спасибо автору за разработку и шрифты. Все супер. Хотелось бы бинарник для внешнего 5102 .

nvalw, 2025-04-01:
Продолжение с эти модулем.
Причем добавления в сток минимальное... Там все это уже было... Добавлен файл с картинками погоды и организован вывод картинки. Из лишнего - шрифт для красивости и поправлен бумбокс показометр...
[image]

SimZs, 2025-05-12:
> Продолжение с этим модулем. Причем добавления в сток минимальное... Там все это уже было... Добавлен файл с картинками погоды и организован вывод картинки. Из лишнего - шрифт для красивости и поправлен бумбокс показометр...
> 
> Чуток причешу очистку зон вывода и выложу все и для всех внешних устройств -РСМ510х, энкодеров(если нужно).
> 
> Я просто не могу выкладывать НЕДОДЕЛКИ, выдавая их за супердостижение... А уж про монетизацию этого процесса было уже смешно... Вы что хотели монетизировать то? Объясните?
> 
> Для нормального отображения шрифтов для других языков я не отказался бы от предоставления коллегами файлов glcdfont и utf8rus, адаптированных для ваших языков. Соответственно в myoptions.h будет сделан автоматический выбор всего этого для вашего языка.
> 
> [image]

> Продолжение с этим модулем. Причем добавления в сток минимальное... Там все это уже было... Добавлен файл с картинками погоды и организован вывод картинки. Из лишнего - шрифт для красивости и поправлен бумбокс показометр...
> 
> Чуток причешу очистку зон вывода и выложу все и для всех внешних устройств -РСМ510х, энкодеров(если нужно).
> 
> Я просто не могу выкладывать НЕДОДЕЛКИ, выдавая их за супердостижение... А уж про монетизацию этого процесса было уже смешно... Вы что хотели монетизировать то? Объясните?
> 
> Для нормального отображения шрифтов для других языков я не отказался бы от предоставления коллегами файлов glcdfont и utf8rus, адаптированных для ваших языков. Соответственно в myoptions.h будет сделан автоматический выбор всего этого для вашего языка.
> 
> [image]

> Продолжение с этим модулем. Причем добавления в сток минимальное... Там все это уже было... Добавлен файл с картинками погоды и организован вывод картинки. Из лишнего - шрифт для красивости и поправлен бумбокс показометр...
> 
> Чуток причешу очистку зон вывода и выложу все и для всех внешних устройств -РСМ510х, энкодеров(если нужно).
> 
> Я просто не могу выкладывать НЕДОДЕЛКИ, выдавая их за супердостижение... А уж про монетизацию этого процесса было уже смешно... Вы что хотели монетизировать то? Объясните?
> 
> Для нормального отображения шрифтов для других языков я не отказался бы от предоставления коллегами файлов glcdfont и utf8rus, адаптированных для ваших языков. Соответственно в myoptions.h будет сделан автоматический выбор всего этого для вашего языка.
> 
> [image]

https://github.com/user-attachments/files/19217294/yoradio_on_JC4827W543C-RUS-0x00000.zip

> Продолжение с этим модулем. Причем добавления в сток минимальное... Там все это уже было... Добавлен файл с картинками погоды и организован вывод картинки. Из лишнего - шрифт для красивости и поправлен бумбокс показометр...
> 
> Чуток причешу очистку зон вывода и выложу все и для всех внешних устройств -РСМ510х, энкодеров(если нужно).
> 
> Я просто не могу выкладывать НЕДОДЕЛКИ, выдавая их за супердостижение... А уж про монетизацию этого процесса было уже смешно... Вы что хотели монетизировать то? Объясните?
> 
> Для нормального отображения шрифтов для других языков я не отказался бы от предоставления коллегами файлов glcdfont и utf8rus, адаптированных для ваших языков. Соответственно в myoptions.h будет сделан автоматический выбор всего этого для вашего языка.
> 
> [image]

Dear nvalw,
Could you please share the Arduino files for this project (modifications). I really like it, I would like to use it for my own radio
Thanks

Alex-art1c, 2025-06-11:
> I have posted a modified version on github with the firmware you provided and my modifications. https://github.com/DaradiciLevente/yoRadio_on_JC4827W543C
> 
> [image]

Thanks. Great mod. Is there any way to add Bluetooth audio output to an external Bluetooth speaker?
It would be great to output audio and controls to an external Bluetooth speaker.

Tanil2008, 2025-08-28:
> > Opublikowałem zmodyfikowaną wersję na github z dostarczonym oprogramowaniem i moimi modyfikacjami. https://github.com/DaradiciLevente/yoRadio_on_JC4827W543C
> > [image]
> 
> Dzięki. Świetny mod. Czy jest jakiś sposób na dodanie wyjścia audio Bluetooth do zewnętrznego głośnika Bluetooth? Byłoby wspaniale wysyłać audio i sterowanie do zewnętrznego głośnika Bluetooth.

Też proszę o informację czy można ten moduł sparować z głośnikiem Bluetooth.

sandyc456, 2025-10-19:
Hello, I kindly request that you provide the latest file from Google Drive from the respected CoolmdXi it's not working on the forum. I can't compile the files from the respected DaradiciLevente I really liked his design. I've read the forum thread several times and seem to have done everything correctly, but I'm still getting compilation errors. There are no problems with the bin files, but I need to make some minor adjustments to suit my needs.