e2002/yoradio #108: setting VS1053 SCI_STATUS for Aliexpress' red board module

morganflint, 2024-11-16:
Hello!
I have yoradio working with a VS1053 module, similar to the second one recommended [here](https://github.com/e2002/yoradio?tab=readme-ov-file#hardware).

I've checked that AVDD (analog part voltage) is connected to the 3.3V regulator output, and that voltage at pin 44 (RCAP) is 1.23V. According to the Recommended Operating Conditions listed on [the datasheet](https://www.vlsi.fi/fileadmin/datasheets/vs1053.pdf), if AVDD is 3.3V, then the recommended setting for reference voltage is 1.65V, as this maximizes the analog output swing. This can be done by changing bit 0 in SCSI_STATUS from 0 to 1.

I've seen that this register is modified in audioVS1053Ex.h (in yoRadio\src\audioVS1053) for enabling VSdsp VU Meter, but I didn't find any other place where I could easily modify bit 0 without disturbing other functionalities. I also don't know if the same issue applies to the other VS1053 recommended modules (green board and Arduino shield).

BTW, this module incorrectly uses 2.5V for CVDD, where the recommended level is 1.8V and the absolute maximum rating is 1.85V. This has already been commented on the 4PDA thread for yoRadio, but I can't find the message now 

PS: I know this project hasn't been updated since some time, but I'm raising this issue just in case it gets active again...

trip5, 2025-02-20:
I just want to thank you for posting this.  I was having some problems with a bunch of VS1053's I got from Aliexpress.  I thought the green boards from a certain seller were reliable.  Ran a multimeter over the RCAP voltage regulator, only to discover the first batch was 1.6ish (correct) but the second batch had an output of 2.5V... one even outputs 3.0V (what the heck).

A whole bunch of experiments before I finally realized this was fixed in v0.7.490 by adding `#define VS_PATCH_ENABLE false` to `myoptions.h`.  No problems now.  That even works on the correct-voltage green boards.  Except of course, for the one that outputs 3.0V.  That one probably has no hope of ever working.

Just putting this here in case someone searches issues before reading the documentation.  Lol.

morganflint, 2025-02-20:
Ok, thank you!
Closing the issue.

closed 2025-02-20 by morganflint