e2002/yoradio #113: Playlist

AndyW999, 2024-11-21:
Please can you let me have an example playlist?

As I don't know the format and Webstations.txt is a little bit out of date.

And not knowing the file format required does not help!

Thanks

Andy.

AndyW999, 2024-11-22:
As I have no playlist cli gives -

##CLI.META#: [connecting]
##AUDIO.INFO#: Connect to new host: "http://"
##AUDIO.INFO#: buffers freed, free Heap: 142588 bytes
##AUDIO.INFO#: Request http:// failed!
##ERROR#:	Request http:// failed!
##ERROR#:	Request http:// failed!
##CLI.META#: ёRadio
##ERROR#:	Error connecting to 
##CLI.META#: Error connecting to 
##ERROR#:	Error connecting to 
##AUDIO.INFO#: buffers freed, free Heap: 142140 bytes
##CLI.META#: Error connecting to 
##CLI.INFO#
##SYS.DATE#: 2024-11-22T16:10:00+03:00
##CLI.NAMESET#: 0 ёRadio
##CLI.VOL#: 77
##CLI.STOPPED#

Please could somebody let me have a playlist so I can actually make this work?

e2002, 2024-11-27:
Hi.
Playlist for example
[playlist.csv](https://github.com/user-attachments/files/17937510/playlist.csv)

closed 2024-11-27 by e2002

aly-fly, 2025-04-06:
> Hi. Playlist for example [playlist.csv](https://github.com/user-attachments/files/17937510/playlist.csv)

@e2002 
Hi, please copy this or similar file to Examples folder.
The KAradio link is broken and this is the only available playlist example.
Alsop please add the format description to Readme.md.

thanks! :)