e2002/yoradio #114: lost pins

r-p0, 2024-11-29:
hello,

i try to setup myoptions.h config with given below hardware, but still get info about lost pins ?
[image]

#ifndef myoptions_h
#define myoptions_h

/* Generated by myoptions.h generator - v0.8.1 */
/* https://e2002.github.io/docs/myoptions-generator.html?r=71,1,2,5,6,76,7,8,31,41,44,47,60,62,64,78,66,67,68,69,70&i=1,2,3,4,18,19,20,21,34,39,48,49,40,41,42,43,44&v=4,5,-1,255,27,25,26,-1,13,255,255,255,2,255,255,255,255 */
/* Put this file next to yoRadio.ino and restart Arduino IDE. */

/* !!! Attention !!! */
/* This configuration depends on some additional libraries. Please install them using the library manager: */
/* Adafruit GFX Library, XPT2046_Touchscreen, RTCLib, IRremoteESP8266 */

/* !!! Do not be alarmed, all right, the default values ​​are not included in this file !!! */

#define DSP_MODEL		DSP_ILI9488
#define I2S_DOUT			255
#define VS1053_CS		27
#define TS_MODEL			TS_MODEL_XPT2046
#define SDC_CS			255
#define RTC_MODULE		DS1307
#define RTC_SDA			255
#define RTC_SCL			255
#define LED_BUILTIN		2

#endif

e2002, 2024-11-29:
Hello,
What modules do you want to connect?
Do you need a SD card? And a touchscreen? RTC module?
If yes, then in the preview window, you can drag the lost pins to yours on the GPIO ESP

r-p0, 2024-11-29:
my ili9488 is built with touchscreen and sd card reader also - so will be nice to use this features
module rtc ds1307 as external clock

to which gpio esp pins need I drag lost pins ?

e2002, 2024-11-29:
Please check out this article: 
https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

e2002, 2024-11-29:
Like this, for example
https://e2002.github.io/docs/myoptions-generator.html?r=71,1,2,5,6,76,7,8,31,41,44,47,60,62,64,78,66,67,68,69,70&i=1,2,3,4,18,19,20,21,34,39,48,49,40,41,42,43,44&v=4,5,-1,21,27,25,26,-1,13,14,17,16,2,35,12,32,33

r-p0, 2024-11-29:
thank you - I'll check now

e2002, 2024-11-30:
- The esp32s3 module is supported only since version 0.9.313b 
- myoptions generator not working for esp32s3 boards. You must assign pins manually.
Please check out this article:
http://wiki.fluidnc.com/en/hardware/ESP32-S3_Pin_Reference

r-p0, 2024-11-30:
Are you planning a version of the myoptions generator for the ESP32S3 board ? Soon ?
Is a compilation error associated with it ? 
What to look for when setting the pins manually for the ESP32S3.

r-p0, 2024-12-25:
Thank you and Merry Christmas to you