e2002/yoradio #115: Add support for ILI9163C

procat2005, 2024-12-01:
You can help for adding this LCD SPI display?
I have one of this and i wait for update on februart 2025. Untill then only ILI9163C i have.
Can you tell me how to add custom display?
Thank you.

procat2005, 2024-12-04:
Hello, can you hep me only this time? It seems that i must search for another gift for the holidays. I have try to add ILI9163C library but i cannot manage to compile by modify defined display to custom. The project use some customized things that i don`t understand. The display is 128x160 but with ST7793 image is mirrored and part of it is overlayed also have some noise. I have try to modify the displayST7735.cpp to match some parameters from ILI9163C library with No success.
I have little time until present time and if i cannot manage to do this i must search for another present for SANTA :) night...
P.S. my wife is cooking only with internet radio, [Радио Борнео] and without it i will eat only prebaked pizza...

procat2005, 2024-12-04:
My display looks like [this[](url)]([image])

procat2005, 2024-12-11:
Solved. Can be closed. It seems that my Arduino IDE is not working as it should. After recompiling in other way it is working right now.

procat2005, 2024-12-13:
[image]

procat2005, 2024-12-13:
[image]

closed 2024-12-13 by procat2005

dorinik, 2024-12-15:
Hello procat2005,
Your build looks really good, congratulations.
I also have a Logitech Squeezebox Radio on which I want to install yoradio.
Can you post some photos of the inside of your radio for inspiration?

Thanks!

procat2005, 2024-12-16:
Hello Dorilink,
I`m ashamed of inside look of my radio, but is working. 
What i have done:
1. ESP32
2. Shield MP3 VS1053 red board
3. PAM 8610 amplifier ( one side used )
4. 2.4 LCD ST7789
5. 12 Volt power supply used for externar HDD ( 12V, 1.5A)
6. 12V to 5V step down converter (https://www.jevoisinc.com/products/step-down-6-24v-in-to-usb-5v-3a-out-dc-dc-converter)
7. Two rotary encoders, extracted from sqeeze box board.
First of all i have unsolder all pins from ESP32 , VS1053, LCD board, PAM amplifier and step down converter have been cleaned for ports and potentiometer. Only boards have been keep.
All boards have been connected with wires, small size have been used. 
LCD board and rotary encoders have been fixed with A+B type of glue (epoxy) in place.
Amplifier, 5 volt converter ESP and VS1053 has been fixed in place on the chassis of box with small screws. VS 1053 has an place where a hole can be made without destroy it. 220 to 12 volt power adapter has a place in the battery compartment.
It is a lot of wires but is working. I will try to open and take a picture.
 Keep in mind: front cover of box is fixed with two screws and it must be slide left to open (about 2 mm)
I have use a filter between VS1053 and PAM amplifier to make MONO sound and reduce noise. It is an divider with 2 x 270 ohm resistors for L and R and 1 x 100 Ohm paralel with 1 x 100nF trough the ground ( minus bar ).
Sqeezer box has 2 speakers. One for mid range and bass, direct connected to PAM amplifier and one tweeter. The tweeter MUST be connected with an 10uF capacitor in SERIES (*). Same way must be wired, in parallel with mid-bass speaker. RESPECT POLARITY. Speaker port has 4 wires blue, green (-) and red, white (+). I remember red and blue is mid-bass and white-green is tweeter.
*IF YOU NOT USE 10uF CAPACITOR, TWEETER WILL BE DESTROYED.
I will come with pictures ASP.

dorinik, 2024-12-16:
Thanks for the detailed answer!
I'm looking forward to the photos too.
You don't have to stress about the connections inside!
The end result is important, and this one is excellent!

procat2005, 2024-12-30:
[image]
[image]
[image]

procat2005, 2024-12-30:
[image]

procat2005, 2024-12-30:
Well, not enough fast but i have some pictures.
You may notice an philips screw in VS1053 board, Next to quartz resonator.
In picture you can see something new, an toggle power switch based on: https://danyk.cz/zap-vyp_en.html as it is using N-FET from briked PC mainboard and on the output i have add an 1K resistor as author specify about capacitive load for this circuit.
Fell free to ask!
I wish to you all: Have a happy new year! 
Thanks to the author of this project! 
I will come back with info about analog buttons for presets memory 1 to 6!

dorinik, 2024-12-31:
Thanks for the photos of the interior!
I look forward to the continuation (I'm curious how you solved the problem with buttons for presets memory)
I think it would be a good idea to post your version of yoradio on your github account, because there are a lot of Squeezebox owners who want to give them a new life!
Thanks again for your work!

A new year full of achievements and lots of work power!

A year full of achievements for the author of this wonderful project too!

procat2005, 2025-01-02:
Have a nice year 2025!
First of all must say that this project is made in very rushing way. I must eat, and my wife does a great job with Russian music in the kitchen. And because we cannot receive radio programs from that far, the only way was internet streams. I am agree that this project can be done more professional, a single PCB board with all stuff but time, time was so short.
May be if i will have another sqeezerbox for spare i will try to do in a different way.
MODS (new one)
1. Adding an 100uF capacitor to the PAM8610 ex-connector for mute. This solves pop noise when the box is turned on.
[image]
2. Adding an 0.1uF capacitor to the reset pin. Increase stability of operation in general.
[image]
3. Adding S1...S6 preset buttons.
3.1 Hardware: 
[image]
3.2 Software:
In controls.cpp from core directory, with Notepad++ add on line 87 like this:
`void initControls() {
  //
  //
  **analogReadResolution(8);**
#if ENC_BTNL!=255
  encoder.begin();
  encoder.setup(readEncoderISR);
  encoder.setBoundaries(0, 254, true);
  encoder.setAcceleration(config.store.encacc);`

next:
on line 135 modify like:
`void loopControls() {
	**int analogValue = analogRead(36);
	if (analogValue > 15) {
	if ((analogValue > 18) && (analogValue < 25)) player.sendCommand({PR_PLAY, 3});
	if ((analogValue > 45) && (analogValue < 65)) player.sendCommand({PR_PLAY, 2});
	if ((analogValue > 85) && (analogValue < 106))player.sendCommand({PR_PLAY, 1});
	if ((analogValue > 115) && (analogValue < 145)) player.sendCommand({PR_PLAY, 4});
	if ((analogValue > 155) && (analogValue < 175)) player.sendCommand({PR_PLAY, 5});
	if ((analogValue > 190) && (analogValue < 220)) player.sendCommand({PR_PLAY, 6});
	}**`
I know is not the best mode of doing this but adding an new variable to define pushed or not pushed button may mess other things and i don`t know how original project have been made. God practice is to reuse an variable predefined in other routine in that way not to interfere those two stuffs. Adding a new variable will reduce amount of RAM already small. Even if you defined boolean, so it will be only one bit.
Pin 36 is used to analog in and you must define in my options (https://e2002.github.io/docs/myoptions-generator.html) some way that pin 36 will not be used by other stuff like encoder button or other things. Leave it unused.

I was use small switches from an Kenwood car cd front end, because those was small enough to be inserted in holes of the S1...S6 buttons. Before fixing them with epoxy i cut all the rubber part of memory buttons, keep only silver shining part.
 I hope you will understand what i wrote upper. Things are not very hard binary processing has not be changed from the beginning of digital computing  until now.  But is hard to solve, even if you have source code, what author have in mind. Messing things are very easy.
[image]
This is original board of the box, notice how hard was to make and how complex it is, CPU, RAM, FLASH, ETHERNET etc. This is what i call electronic waste. And some factories are the best in this. Waste of time and work to obtain an thing that is working for one summer. Waste factories.
Oki. Have fun with YORadio!

dorinik, 2025-01-14:
Hello procat2005,
Please post your settings from myoptions.h.
Thanks!

procat2005, 2025-01-15:
#ifndef myoptions_h
#define myoptions_h

/* Generated by myoptions.h generator - v0.8.1 */
/* https://e2002.github.io/docs/myoptions-generator.html?r=71,2,6,7,19,41,44,47,64,67&i=1,2,3,4,18,19,20,21,39,41&v=4,5,-1,255,27,25,26,-1,22,21 */
/* Put this file next to yoRadio.ino and restart Arduino IDE. */

/* !!! Attention !!! */
/* This configuration depends on some additional libraries. Please install them using the library manager: */
/* Adafruit GFX Library, Adafruit ILI9341, IRremoteESP8266 */

/* !!! Do not be alarmed, all right, the default values ​​are not included in this file !!! */

#define DSP_MODEL			DSP_ST7789
#define I2S_DOUT			255
#define VS1053_CS			27
#define BTN_LEFT      17
#define BTN_CENTER    16
#define BTN_RIGHT     2
#define BTN_MODE      15
#define ENC_BTNR      14
#define ENC_BTNL      13
#define ENC_BTNB      12
#define ENC2_BTNR     35
#define ENC2_BTNL     33
#define ENC2_BTNB     32
#define SDC_CS			22
#define IR_PIN			21

#endif

dorinik, 2025-01-16:
Hello procat2005,
I tested your changes in the code for presets and it works, except for key 3, where the code is also different.
What rule did you apply to get the minimum and maximum analogValue?
Can the code be extended to get more than 6 presets (for example 12)?
Thanks for the advice!
[image]
[image]

procat2005, 2025-01-17:
Sorry, is about bad formatting of github.
First picture is how is about to look the readings, is correct formating in compiler.
Second picture is how you can read analog values by serial print by putty console, if you want to find what values is about each button.

[image]

[image]
If you use, temporary i mean, the second code you be reading in serial console, by wire (on USB connection connect with PUTTY on serial port "XX" with 115200bauds) you will read the analog value of pin.
Make schematic with 12 buttons, compile with second picture code, turn on and read all values of buttons by console, write them on paper.
Modify code like first picture with new values and new commands by adding them.
Each line contain an interval of analog value between minimum and maximum value.
If analog value is between those limimits is executed the command.
First line skip executing if analogvalue is below 15 (if no button is pressed)
Limits is upon you. An interval of +/- 10 units is ok i think.
You can use for all kind of commands , not only for presets.

dorinik, 2025-01-17:
Thanks for the clarifications and explanations!
It was only necessary to change if (analogValue > 15) to if (analogValue > 5), because the first button did not work. I think that, to work with if (analogValue > 15), the first resistor from GND - 1K, should be replaced with a 2K one.
Now everything works perfectly!
It works well with 12KEY.
I will do tests with ESP32-S3.

The missing space at key1 has been corrected - you can see it in photo2 I posted yesterday.

Thanks again for the help!