e2002/yoradio #116: ST7789 x240 does not display

RecepAGulec, 2024-12-01:
Hi,
I have an ST7789 display without CS and MISO pins. (240x240). I tested it on 8266 and it works. But it doesn't work with yoRadio. I also tried it as "CS 255" in myoptions.h file.
Is there another setting item for this display that I don't know about?

e2002, 2024-12-01:
https://github.com/e2002/yoradio/wiki/Available-display-models#2-dsp_st7789_240

ST7789 requires further development when used in conjunction with the VS1053 module
See the link https://www.instructables.com/Adding-CS-Pin-to-13-LCD/

RecepAGulec, 2024-12-01:
You stated that "the ST7789 needs further development when used with the VS1053 module."
I use the PCM5102, not the VS1053 module.
Is it absolutely necessary to add CS to the ST7789 display?

e2002, 2024-12-01:
The Adafruit-ST7735 library comes with an example called graphicstest_st7789
https://github.com/adafruit/Adafruit-ST7735-Library/tree/master/examples/graphicstest_st7789
Does it work with your display?

RecepAGulec, 2024-12-01:
The sketch located at https://github.com/adafruit/Adafruit-ST7735-Library/tree/master/examples/graphicstest_st7789 did not work on my ST7789 display. 
However, Adafruit's sketch, adapted only to ESP8266 and available at https://simple-circuit.com/esp8266-nodemcu-st7789-tft-ips-display/, worked perfectly.

e2002, 2024-12-02:
The project uses the Adafruit-ST7735 library to work with the ST7789 display.
If the example provided with the library does not work, it will not work in the project either.
Please double-check that the display is connected correctly and ensure that the example from the library runs successfully.
If this does not help, modify the CS pin as described in the link above.
Unfortunately, this is all the assistance I can provide.