e2002/yoradio #119: ESP Wrover +ST7796 +VS1053

JureqB, 2024-12-16:
Everything compiles correctly, uploads, but after starting play, the UV indicator turns on to maximum, the message "CONNECTING" is displayed and there is no sound at the output.
[image]

On the port monitor it looks like everything is working fine:

13:20:00.536 -> ets Jun  8 2016 00:22:57
13:20:00.536 -> 
13:20:00.536 -> rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
13:20:00.536 -> configsip: 0, SPIWP:0xee
13:20:00.536 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
13:20:00.536 -> mode:DIO, clock div:1
13:20:00.536 -> load:0x3fff0030,len:1344
13:20:00.536 -> load:0x40078000,len:13964
13:20:00.536 -> load:0x40080400,len:3600
13:20:00.536 -> entry 0x400805f0
13:20:01.438 -> ##[BOOT]#	************************************************
13:20:01.438 -> ##[BOOT]#	*               ёPadio v0.9.370                *
13:20:01.438 -> ##[BOOT]#	************************************************
13:20:01.438 -> ##[BOOT]#	------------------------------------------------
13:20:01.485 -> ##[BOOT]#	arduino:	10819
13:20:01.485 -> ##[BOOT]#	compiler:	8.4.0
13:20:01.485 -> ##[BOOT]#	esp32core:	2.0.17
13:20:01.485 -> ##[BOOT]#	chip:		model: ESP32-D0WDQ6 | rev: 1 | id: 6017152 | cores: 2 | psram: 4192123
13:20:01.485 -> ##[BOOT]#	display:	19
13:20:01.485 -> ##[BOOT]#	audio:		VS1053 (27, 25, 26, 33, false)
13:20:01.485 -> ##[BOOT]#	audioinfo:	false
13:20:01.485 -> ##[BOOT]#	smartstart:	0
13:20:01.485 -> ##[BOOT]#	vumeter:	false
13:20:01.485 -> ##[BOOT]#	softapdelay:	0
13:20:01.485 -> ##[BOOT]#	flipscreen:	false
13:20:01.485 -> ##[BOOT]#	invertdisplay:	false
13:20:01.485 -> ##[BOOT]#	showweather:	false
13:20:01.485 -> ##[BOOT]#	buttons:	left=255, center=255, right=255, up=255, down=255, mode=255, pullup=true
13:20:01.532 -> ##[BOOT]#	encoders:	l1=36, b1=39, r1=34, pullup=false, l2=255, b2=255, r2=255, pullup=true
13:20:01.532 -> ##[BOOT]#	ir:		35
13:20:01.532 -> ##[BOOT]#	------------------------------------------------
13:20:01.532 -> ##[BOOT]#	CONFIG_VERSION	1
13:20:01.532 -> ##[BOOT]#	SPIFFS mounted
13:20:01.532 -> ##[BOOT]#	display.init	done
13:20:02.249 -> ##[BOOT]#	player.init	done
13:20:04.046 -> ##[BOOT]#	network.begin
13:20:04.142 -> ##[BOOT]#	Attempt to connect to siatka
13:20:04.142 -> ##[BOOT]#	..
13:20:04.615 -> ##[BOOT]#	done
13:20:04.662 -> ##[BOOT]#	netserver.begin	done
13:20:04.662 -> ##[BOOT]#	telnet.begin	done
13:20:04.662 -> ##[BOOT]#
13:20:04.662 -> ##[BOOT]#	Ready! Go to http:/10.0.0.161/ to configure
13:20:04.662 -> ##[BOOT]#	------------------------------------------------
13:20:04.662 -> ##[BOOT]#
13:20:04.756 -> ##CLI.META#: [ready]
13:20:05.374 -> ##CLI.META#: [connecting]
13:20:05.374 -> ##AUDIO.INFO#: Connect to new host: "http://n13a-eu.rcs.revma.com/ye5kghkgcm0uv?rj"
13:20:05.374 -> ##AUDIO.INFO#: Song stopped correctly after 0 msec
13:20:05.469 -> ##AUDIO.INFO#: Connection has been established in 91 ms, free Heap: 168656 bytes
13:20:05.564 -> ##AUDIO.INFO#: redirect to new host "http://n13a-eu.rcs.revma.com/ye5kghkgcm0uv?rj=&rj-tok=AAABk89ybqgAFNcJyAPmgMZLVw&rj-ttl=5"
13:20:05.564 -> ##AUDIO.INFO#: Connect to new host: "http://n13a-eu.rcs.revma.com/ye5kghkgcm0uv?rj=&rj-tok=AAABk89ybqgAFNcJyAPmgMZLVw&rj-ttl=5"
13:20:05.612 -> ##AUDIO.INFO#: Song stopped correctly after 0 msec
13:20:05.659 -> ##AUDIO.INFO#: Connection has been established in 51 ms, free Heap: 163512 bytes
13:20:05.659 -> ##CLI.INFO#
13:20:05.659 -> ##SYS.DATE#: 1970-01-01T01:00:04+03:00
13:20:05.659 -> ##CLI.NAMESET#: 1 Radio 357
13:20:05.659 -> ##CLI.META#: [connecting]
13:20:05.659 -> ##CLI.VOL#: 191
13:20:05.659 -> ##CLI.PLAYING#
13:20:05.706 -> ##AUDIO.INFO#: format is aac
13:20:05.706 -> ##AUDIO.INFO#: icy-name: Radio 357
13:20:07.984 -> ##WEATHER###: description: zachmurzenie duże, temp:9.1 C, pressure:988mmHg, humidity:93%

e2002, 2024-12-16:
Nie widzę problemów z odtwarzaniem tej stacji na VS1053. Być może masz problem sprzętowy?

```
15:47:02.759 -> ##AUDIO.INFO#: Connect to new host: "http://n13a-eu.rcs.revma.com/ye5kghkgcm0uv?rj"
15:47:03.356 -> ##AUDIO.INFO#: Song stopped correctly after 20 msec
15:47:03.489 -> ##AUDIO.INFO#: Connection has been established in 152 ms, free Heap: 180984 bytes
15:47:03.588 -> ##AUDIO.INFO#: redirect to new host "http://n02a-eu.rcs.revma.com/ye5kghkgcm0uv?rj=&rj-tok=AAABk8-LJx8AVBBv2qpo82HsDA&rj-ttl=5"
15:47:03.588 -> ##AUDIO.INFO#: Connect to new host: "http://n02a-eu.rcs.revma.com/ye5kghkgcm0uv?rj=&rj-tok=AAABk8-LJx8AVBBv2qpo82HsDA&rj-ttl=5"
15:47:03.621 -> ##AUDIO.INFO#: Song stopped correctly after 0 msec
15:47:03.754 -> ##AUDIO.INFO#: Connection has been established in 131 ms, free Heap: 180848 bytes
15:47:03.854 -> ##AUDIO.INFO#: format is aac
15:47:03.854 -> ##AUDIO.INFO#: icy-name: Radio 357
15:47:04.219 -> ##AUDIO.INFO#: stream ready
15:47:04.219 -> ##AUDIO.INFO#: buffer filled in 377 ms
15:47:04.451 -> ##AUDIO.INFO#: StreamTitle='Stevie Wonder - Superstition'
```

[image]

JureqB, 2024-12-16:
Wiem że stacja działa bo ten sam moduł VS1053 w innej konfiguracji działa mi poprawnie.

JureqB, 2024-12-17:
Może to da jakąś podpowiedź, podminiełem VS1053 na identyczny ale  bez zworki na nóżkach 33 i 34 i sytuacja się zmieniła o tyle że UV na startuje nafull od razu tylko po kiku sekundach, przy próbie zmienay stacji (przez www) jest dług aprzerwa zanim zareaguje a sytuacja na porcie wygląda tak: 

12:00:08.258 -> ets Jun  8 2016 00:22:57
12:00:08.258 -> 
12:00:08.258 -> rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
12:00:08.258 -> configsip: 0, SPIWP:0xee
12:00:08.258 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
12:00:08.258 -> mode:DIO, clock div:1
12:00:08.258 -> load:0x3fff0030,len:4832
12:00:08.258 -> load:0x40078000,len:16460
12:00:08.258 -> load:0x40080400,len:4
12:00:08.258 -> load:0x40080404,len:3504
12:00:08.258 -> entry 0x400805cc
12:00:09.208 -> ##[BOOT]#	************************************************
12:00:09.208 -> ##[BOOT]#	*               ёPadio v0.9.373                *
12:00:09.256 -> ##[BOOT]#	************************************************
12:00:09.256 -> ##[BOOT]#	------------------------------------------------
12:00:09.256 -> ##[BOOT]#	arduino:	10819
12:00:09.256 -> ##[BOOT]#	compiler:	12.2.0
12:00:09.256 -> ##[BOOT]#	esp32core:	3.0.7
12:00:09.256 -> ##[BOOT]#	chip:		model: ESP32-D0WDQ6 | rev: 100 | id: 6017152 | cores: 2 | psram: 4194304
12:00:09.256 -> ##[BOOT]#	display:	19
12:00:09.256 -> ##[BOOT]#	audio:		VS1053 (27, 25, 26, -1, false)
12:00:09.256 -> ##[BOOT]#	audioinfo:	false
12:00:09.256 -> ##[BOOT]#	smartstart:	0
12:00:09.256 -> ##[BOOT]#	vumeter:	false
12:00:09.256 -> ##[BOOT]#	softapdelay:	0
12:00:09.256 -> ##[BOOT]#	flipscreen:	false
12:00:09.256 -> ##[BOOT]#	invertdisplay:	false
12:00:09.303 -> ##[BOOT]#	showweather:	false
12:00:09.303 -> ##[BOOT]#	buttons:	left=255, center=255, right=255, up=255, down=255, mode=255, pullup=true
12:00:09.303 -> ##[BOOT]#	encoders:	l1=34, b1=39, r1=36, pullup=false, l2=255, b2=255, r2=255, pullup=true
12:00:09.303 -> ##[BOOT]#	ir:		35
12:00:09.303 -> ##[BOOT]#	------------------------------------------------
12:00:09.303 -> ##[BOOT]#	CONFIG_VERSION	1
12:00:09.303 -> ##[BOOT]#	SPIFFS mounted
12:00:09.303 -> ##[BOOT]#	display.init	done
12:00:09.967 -> ##[BOOT]#	player.init	done
12:00:11.675 -> ##[BOOT]#	network.begin
12:00:11.768 -> ##[BOOT]#	Attempt to connect to siatka
12:00:11.768 -> ##[BOOT]#	..
12:00:12.242 -> ##[BOOT]#	done
12:00:12.290 -> ##[BOOT]#	netserver.begin	done
12:00:12.290 -> ##[BOOT]#	telnet.begin	done
12:00:12.290 -> ##[BOOT]#
12:00:12.290 -> ##[BOOT]#	Ready! Go to http:/10.0.0.161/ to configure
12:00:12.290 -> ##[BOOT]#	------------------------------------------------
12:00:12.290 -> ##[BOOT]#
12:00:12.386 -> ##CLI.META#: [ready]
12:00:12.953 -> ##CLI.META#: [connecting]
12:00:13.000 -> ##AUDIO.INFO#: Connect to new host: "http://n13a-eu.rcs.revma.com/ye5kghkgcm0uv?rj"
12:00:14.992 -> ##AUDIO.INFO#: Song stopped incorrectly!
12:00:14.992 -> ##AUDIO.INFO#: after song stopped incorrectly
12:00:14.992 -> ##AUDIO.INFO#: REG            dec      bin               hex
12:00:15.040 -> ##AUDIO.INFO#: -----------  -------  ----------------  -------
12:00:15.040 -> ##AUDIO.INFO#: MODE          02056   0000100000001000   0808
12:00:15.040 -> ##AUDIO.INFO#: STATUS        00576   0000001001000000   0240
12:00:15.040 -> ##AUDIO.INFO#: BASS          61967   1111001000001111   F20F
12:00:15.040 -> ##AUDIO.INFO#: CLOCKF        24576   0110000000000000   6000
12:00:15.040 -> ##AUDIO.INFO#: DECODE_TIME   00003   0000000000000011   0003
12:00:15.040 -> ##AUDIO.INFO#: AUDATA        44101   1010110001000101   AC45
12:00:15.040 -> ##AUDIO.INFO#: WRAM          60004   0100100010110001   FFFF
12:00:15.040 -> ##AUDIO.INFO#: WRAMADDR      22528   0101100000000000   5800
12:00:15.087 -> ##AUDIO.INFO#: HDAT0         00000   0000000000000000   0000
12:00:15.087 -> ##AUDIO.INFO#: HDAT1         00000   0000000000000000   0000
12:00:15.087 -> ##AUDIO.INFO#: AIADDR        00000   0000000000000000   0000
12:00:15.087 -> ##AUDIO.INFO#: VOL           02827   0000101100001011   0B0B
12:00:15.087 -> ##AUDIO.INFO#: AICTRL0       00000   0000000000000000   0000
12:00:15.087 -> ##AUDIO.INFO#: AICTRL1       00000   0000000000000000   0000
12:00:15.087 -> ##AUDIO.INFO#: AICTRL2       00000   0000000000000000   0000
[... 9 lines cut ...]
12:00:15.275 -> ##AUDIO.INFO#: Connect to new host: "http://n42a-eu.rcs.revma.com/ye5kghkgcm0uv?rj=&rj-tok=AAABk9RPuroAa7Q31zC1i0jhVQ&rj-ttl=5"
12:00:15.418 -> ##WEATHER###: description: zachmurzenie duże, temp:8.1 C, pressure:990mmHg, humidity:93%
12:00:17.262 -> ##AUDIO.INFO#: Song stopped incorrectly!
12:00:17.262 -> ##AUDIO.INFO#: after song stopped incorrectly
12:00:17.262 -> ##AUDIO.INFO#: REG            dec      bin               hex
12:00:17.309 -> ##AUDIO.INFO#: -----------  -------  ----------------  -------
12:00:17.309 -> ##AUDIO.INFO#: MODE          02056   0000100000001000   0808
12:00:17.309 -> ##AUDIO.INFO#: STATUS        00576   0000001001000000   0240
12:00:17.309 -> ##AUDIO.INFO#: BASS          61967   1111001000001111   F20F
12:00:17.309 -> ##AUDIO.INFO#: CLOCKF        24576   0110000000000000   6000
12:00:17.309 -> ##AUDIO.INFO#: DECODE_TIME   00005   0000000000000101   0005
12:00:17.309 -> ##AUDIO.INFO#: AUDATA        44101   1010110001000101   AC45
12:00:17.309 -> ##AUDIO.INFO#: WRAM          10897   0110000001001010   E8A0
12:00:17.309 -> ##AUDIO.INFO#: WRAMADDR      22528   0101100000000000   5800
[... cut ...]
12:00:17.356 -> ##AUDIO.INFO#: AICTRL0       00000   0000000000000000   0000
12:00:17.356 -> ##AUDIO.INFO#: AICTRL1       00000   0000000000000000   0000
12:00:17.356 -> ##AUDIO.INFO#: AICTRL2       00000   0000000000000000   0000
12:00:17.356 -> ##AUDIO.INFO#: AICTRL3       00000   0000000000000000   0000
12:00:17.448 -> ##AUDIO.INFO#: Connection has been established in 68 ms, free Heap: 164504 bytes
12:00:17.495 -> ##AUDIO.INFO#: format is aac
12:00:17.495 -> ##AUDIO.INFO#: icy-name: Radio 357

aleks434, 2025-02-17:
похожая ситуация. индикатор уровня в максимуме. плата проверена на другом устройстве, исправна. прошивку три раза перешивал с разными ногами подключения. есть заворот от микрофона. т.е. свист в колонке. проблема в правильности прошивки похоже.вклучен режим работы от микрофона встроенного на плате.

MuxauJl-RU, 2025-02-18:
Аналогичная ситуация. Из наблюдений:
1. Про то, что не работает звук.
Дисплей с контроллером, как и у автора темы. Использовал две разные версии плат с VS1053B. Дело оказалось в модуле ESP32-WROVER. Обратил внимание, что ревизия WROOVER-a, которую использую, похоже, как и у автора темы, старая, помеченная на сайте производителя, как EOL. Отказавшись от использования PSRAM, радио заиграло. После отключения PSRAM, лог такой:

##[BOOT]#	************************************************
##[BOOT]#	*               ёPadio v0.9.434                *
##[BOOT]#	************************************************
##[BOOT]#	------------------------------------------------
##[BOOT]#	arduino:	10819
##[BOOT]#	compiler:	12.2.0
##[BOOT]#	esp32core:	3.0.7
##[BOOT]#	chip:		model: ESP32-D0WDQ6 | rev: 100 | id: 7720784 | cores: 2 | psram: 0

Если есть способ заставить работать PSRAM на этом модуле - замечательно!

2. VU метр всё-равно всегда показывает максимум, подрагивая верхней частью верхнего "кирпичика" в обоих каналах.

aleks434, 2025-02-18:
в телеграмм  https://t.me/YoRadio_Trepalka там свою проблему решить помогли, я рисунок выложил.

https://t.me/YoRadio_Trepalka/12374

MuxauJl-RU, 2025-02-19:
VU метр заработал после удаления из myoptions ранее внесённого туда, во время попыток заставить это хоть как работать, запрета на записывание FLAC плагина для VS1053B. Индикация VU метра не синхронизирована со звуком, но этот недуг тоже уже видел у кого-то...

Осталось решить основную проблему - как заставить этот проект работать на старом WROVER-е (как у автора темы), у которого похоже, что не 8 Мб, а 4 Мб PSRAM (https://4pda.to/forum/index.php?showtopic=1010378&st=820#entry107440760):

...
12:00:09.256 -> ##[BOOT]# chip: model: ESP32-D0WDQ6 | rev: 100 | id: 6017152 | cores: 2 | psram: 4194304
...

У меня такие же значения для PSRAM выводит, если прошить проект, как для WROVER.

P.S. С подключением WROVER-а всё в порядке, выводы 16, 17 и 0 не задействованы для проекта, выводы энкодера подтянуты внешними резисторами.