e2002/yoradio #121: Since version 0.9.350 the second subtitle does not work.

hevet, 2024-12-16:
When some stations are not broadcasting a song, but are conducting a conversation, for example, they display a second subtitle. Since version 0.9.350, this second subtitle is no longer displayed. When a song is broadcast, the artist and title are displayed as before in the second line. Below are photos from two different radios, but they show what I mean. On the radio with a color display, version 0.9.300 is used, here the second subtitle works, on the other with a yellow display, version 0.9.350 is used and here it no longer works.
[image]
[image]

e2002, 2024-12-16:
If station titles contain diacritical characters, only the station name is displayed in the first line instead of the full title.
This is a feature of how fonts work in the AdafruitGFX library.
Titles without diacritical characters are displayed across two lines in the latest versions.

[image]

hevet, 2024-12-16:
Hmm. There are no accents in the second subtitle. I also set the station name without accents and it still does not display the second subtitle. There are two lines of information for the song, while for example there is only one for the message, in the older version 0.9.300 on the same version of the AdafruitGFX library, the second subtitle works.

hevet, 2024-12-16:
##AUDIO.INFO#: redirect to new host "https://n31a-eu.rcs.revma.com/ypqt40u0x1zuv?rj-ttl=5&rj-tok=AAABk9ClmEAAfCDgiLpoBq35CQ"
##AUDIO.INFO#: Connect to new host: "https://n31a-eu.rcs.revma.com/ypqt40u0x1zuv?rj-ttl=5&rj-tok=AAABk9ClmEAAfCDgiLpoBq35CQ"
##AUDIO.INFO#: Song stopped correctly after 0 msec
##AUDIO.INFO#: SSL has been established in 683 ms, free Heap: 132528 bytes
##AUDIO.INFO#: format is mp3
##AUDIO.INFO#: icy-name: Radio Nowy Swiat
##AUDIO.INFO#: stream ready
##AUDIO.INFO#: StreamTitle='Radio Nowy Świat - Pion i poziom!'
##CLI.META#: Radio Nowy Świat

closed 2024-12-17 by hevet