e2002/yoradio #122: Reconnect stream automatically

ciprisat, 2024-12-20:
Hi , the radio stream is not automatically reconnected when the internet is restored. can it be solved? Thank you.

ciprisat, 2024-12-23:
a loop check of the player can be implemented script if it is in connected status permanent

ciprisat, 2025-01-02:
1.Implemented webinterface select on port remote acces. Ex. 192.168.0.160:5000
2. Reconnect automatically lost router no ap mode
3. Reconnect automatically stream radio playlist error connect. check in the loop if it is connected
Thank you very much

ciprisat, 2025-01-05:
create my personalized radio code offer price dollars

ciprisat, 2025-01-06:
Contact 
ciprian.bobar.cb@gmail.com

e2002, 2025-09-16:
Hello!
In the latest versions, there are no problems with automatic playback when the Internet connection is restored.