e2002/yoradio #123: ESP32-S3 - simple myoptions.h generator

trip5, 2024-12-23:
I had some trouble getting the VS1053 attached to the ESP32-S3... I did discover in the `s3_pins_arduino.h` are these defaults (`VSPI`):
```
static const uint8_t TX = 43;
static const uint8_t RX = 44;

static const uint8_t SDA = 8;
static const uint8_t SCL = 9;

static const uint8_t SS    = 10;
static const uint8_t MOSI  = 11;
static const uint8_t MISO  = 13;
static const uint8_t SCK   = 12;
```
Used `MOSI`, `MISO`, and `SCK` for the VS1053 and it works well!

By the way using `#define VS_HSPI       true` will get:
```
[   253][E][esp32-hal-spi.c:178] spiAttachSCK(): HSPI Does not have default pins on ESP32S3!
[   262][E][esp32-hal-spi.c:215] spiAttachMISO(): HSPI Does not have default pins on ESP32S3!
[   270][E][esp32-hal-spi.c:254] spiAttachMOSI(): HSPI Does not have default pins on ESP32S3!
```
So yeah, unless you want to open up the code to use custom pins...? but you shouldn't need to.  I've got some S3-Zero boards and pins 8-13 are all exposed there too (and it's smaller than a D1).  These pins seem very standard.

I did some work for another radio project and some pretty heavy research into what pins are unusable and/or reserved on the ESP32-S3 if I can share here which may be useful for the auto-configurator.  Many of these pins are exposed too but essentially unusable for a project like this...

```
Pin 0 // Strapping / BOOT pin (possible but not recommended)
Pin 3 // JTAG strapping pin (possible but not recommended)
Pin 19 // Reserved USB D-
Pin 20 // Reserved USB D+
Pins 22-25 // Not exposed on board / Invalid pin
Pins 26-32 // Reserved for SPI Flash
Pins 33-34 // Not exposed on board (will cause crash if added)
Pins 35-37 // Reserved for PSRAM if 8GB or more (usually exposed but can cause crash if used)
Pin 43 // Reserved USB/Serial U0TXD (possible but not recommended)
Pin 44 // Reserved USB/Serial U0RXD (possible but not recommended)
Pin 45 // Reserved SPI Flash voltage
Pin 46 // Reserved Boot mode
```
The pins go up to (including) 48.

So yeah, just opening this as an 'issue' to help out for the next version of the auto config!  Thanks again for making such a great (and beautiful UI) project!

So for anyone else looking for some configuration... I used this in my `myoptions.h` file:
```
#define I2S_DOUT   255
#define VS_HSPI   false
#define VS1053_CS   5
#define VS1053_DCS   4
#define VS1053_DREQ   21
#define VS1053_RST   -1    /* set to -1 if connected to Esp EN pin */
```
but of course you should be able to use whatever pins you like for `CS`, `DCS`, and `DREQ`.

trip5, 2025-01-05:
Putting this here - still testing but: https://trip5.github.io/yoradio-docs/docs/myoptions-generator.html

aly-fly, 2025-04-06:
I recommend one more update to the generator.
I2C OLEDs also have RST pin, on many boards it's connected to a GPIO, also `myoptions.h` file allows setting it. But it's missing in the generator.
`#define I2C_RST             16`

aly-fly, 2025-04-06:
IK would suggest adding some CYD modules to the generator, for example this one is quite complex to set-up, but it would be great to have it automated in the Board selector: 
https://github.com/e2002/yoradio/issues?q=cyd

trip5, 2026-05-14:
Bumping this, as I won't be adding to it any further.
https://trip5.github.io/ehRadio_myoptions/generator.html
https://github.com/trip5/ehRadio_myoptions
https://github.com/e2002/docs/pull/2